RaspAP/raspap-webgui #245: RASPAP WiFI Portal Wireless Scan doesn't work when manual install method is used

ionicbond, 2018-10-08:
### Subject of the issue
I can not scan for wireless networks via the RASPAP WIFi Portal after completing the MANUAL INSTALL steps.  However, when I use the Quick Install method, I can then scan for wireless networks via the RASPAP Wifi Portal.  I believe this shows clearly that something is missing from the manual install that is included with the Quick Install.

### Your environment
* Raspberry Pi hardware:  Pi 3 Model B 
* Raspbian version:  VERSION="8 (jessie)"
* RaspAP Quick Install or Manual setup?  MANUAL SETUP
* Other software or services running with RaspAP?  NONE

### Steps to reproduce
Install Raspian Jessie Lite image, go through the manual install.  Then try to scan for wireless networks via the RASPAP WiFi Portal.

### Expected behaviour
 I should be able to scan for wireless networks via the RASPAP Wifi Portal after I complete the MANUAL INSTALL steps.  However, I can only scan for wireless networks if I complete the QUICK INSTALL method.

### Actual behaviour
When I go through the manual install steps and I try to scan for wireless networks using the RASPAP WIFI Portal, nothing happens and no wireless networks show up EXCEPT for those that are already listed in my /etc/wpa_supplicant/wpa_supplicant.conf file.  I should see at least an additional 8-10 wireless networks

PRO2XY, 2018-11-01:
I have observed the same issue. Running `sudo iwlist wlan0 scan | grep ESSID` on the raspberry pi does return the scanned SSIDs, so it is some issue with RaspAP itself.

PRO2XY, 2018-11-03:
The issue is with incorrect / incomplete entry in `/etc/sudoers`
You need to add the interface option to the /sbin/wpa_cli commands.
Simply change 
```
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli scan_results
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli scan
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli reconfigure
```
to
```
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli -i wlan0 scan_results
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli -i wlan0 scan
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli -i wlan0 reconfigure
```
Replace `wlan0` with your wifi interface name if required.

PRO2XY, 2018-11-03:
You may also need to change `$scanned_networks` to `$networks` in [/includes/configure_client.php at line 9](https://github.com/billz/raspap-webgui/blob/c366ef0abbe31f3999a751f2d532a1f5c294407a/includes/configure_client.php#L9)

From what I can figure, it is declared as $scanned_networks on that line, but used as $networks everywhere else.

commit d912eb5a (2018-11-08): Update manual installation steps

Resolves #245
closed 2018-11-08 by billz