RaspAP/raspap-webgui #246: Upgrade steps/procedure

benzino77, 2018-10-09:
### Subject of the issue
How to upgrade from 1.3.0 to 1.3.1 to not loose the configuration?

### Your environment
* Raspberry Pi hardware: Pi 3 Model B
* Raspbian version: Stretch 9.4
* RaspAP Quick Install or Manual setup? Quick install
* Other software or services running with RaspAP? mosquitto

billz, 2018-10-11:
Excellent question. Upgrading your current install to a **specific release** is very straightforward. From the RaspAP directory, enter the following:
```
sudo git fetch --tags
sudo git checkout tags/1.3.1
```
A tag is a pointer that isn't connected to the main development tree that git knows about. As a result, git will reply that you're in a 'detached HEAD' state. This isn't a big deal, it just means that you have a specific version of the code that isn't connected to the git tree. 

If you want the latest **bleeding edge** commits from the master branch, use the following:
```
sudo git checkout master
sudo git pull origin master
```
Whichever method you choose (specific release or latest updates), your RaspAP configuration won't be changed.

benzino77, 2018-10-12:
Thanks Bill!

closed 2018-10-12 by benzino77

benzino77, 2018-10-12:
After upgrade there is still information in the left upper corner that the version is 1.3.0.

billz, 2018-10-12:
Yes, this was fixed by a commit after 1.3.1 😉You can verify the current release by checking the first line of `includes/config.php`

```
define('RASPI_VERSION', '1.3.1');
```

seebz, 2018-10-16:
> [...] From the RaspAP directory, [...]
It is the `/var/www/html` directory, right ?

I think It would be good to specify it in the FAQ for future users :)

billz, 2018-10-16:
@seebz that's correct. updated the FAQ accordingly, thanks