RaspAP/raspap-webgui #247: 2 Questions

MrTob, 2018-10-10:
Hallo,
erstmal Danke Für das super Programm. Ich hätte zwei Fragen
Als erstes kann ich den den WLAN Hotspot auf "hidden" stellen?
Mein zweite Frage wäre kann ich ohne Bedenken PhP und apache und MySQL am PI installieren ohne dass es dein Programm stört

D9ping, 2018-10-10:
_For two questions it is better to open two issues separate issues._

It currently not possible via the web interface to set a empty SSID broadcast (hidden SSID),
But you can configure hostapd manually to do this by adding the following line:
ignore_broadcast_ssid=1
to the /etc/hostapd/hostapd.conf file.
Then restart hostapd with:
sudo systemctl restart hostapd.service
And you will have a hotspot with a hidden SSID. 
Some client cannot deal with empty length SSID broadcast and they will fail to connect.
As a security measure it's pretty useless, as your trusted devices connecting to the hotspot send the SSID in clear for anyone wanting to also know. Hiding the SSID is 'security through obscurity'.

About your second question.
You cannot run lighttpd and apache on the same port 80(http) as only one program can bind to port 80 on the same device.
You can run multiple virtual hosts("vhost" for short) on lighttp or on apache to run multiple website's incl. RaspAP on the same device and port. But this configuration take some effort and time to set-up correctly, Including setting up your DNS. Currently the RaspAP is configured as the default virtual host for port 80 for lighttp making it serve always the RaspAP webinterface on port 80 by default (unless any other vhost is configure and that matches the host http 1.1>= header).

billz, 2018-10-14:
Hidden SSIDs are now supported (thanks @D9ping). 
The second issue is slightly more involved and beyond the scope of this project.

billz, 2018-10-14:
@MrTob just to clarify your second question, changing lighttpd's default server port is trivial and should let you run another webserver alongside it without major issues. Edit `/etc/lighttpd/lighttpd.conf` and change the following line:
```
server.port                 = 8080 # default is 80
```
then give the service a kick...
```
sudo service lighttpd restart
```
You can then access RaspAP as before with the new port number in the URI. eg., http://raspberrypi.local:8080. This was requested in #146. It's been added to a (growing!) todo list.

closed 2019-03-04 by billz