RaspAP/raspap-webgui #252: RasPap - RPI 3B+ - 5GHZ Channels/802.11ac

Juln75, 2018-10-14:
### Subject of the issue
Hi @billz ,
I've read that RasPap now support 5Ghz channels but I can't find how to use this feature.
I'm missing something ?

Issue : #https://github.com/billz/raspap-webgui/pull/161#issuecomment-368297643

> Enhancements
> Added support for 5GHz channels. Priority now preserved in wpa_supplicant.conf (thanks @rmarshall31).

### Your environment
* Raspberry Pi 3 B+
* Raspbian Strech
* RaspAP Quick Install (1.3.1)
* Other software or services running with RaspAP: None

### Steps to reproduce
Do a fresh install on a RPI 3B+
Go on administration panel (http://10.3.141.1/) -> Configure hotspot
Basic, Wireless mode -> ac is missing ?
https://en.wikipedia.org/wiki/IEEE_802.11ac
[image]

### Expected behaviour
add 5GHz IEEE 802.11.b/g/n/**ac** wireless LAN

### Actual behaviour
Wireless mode "ac" is not supported ?

Thank you in advance,
J

D9ping, 2018-10-14:
Duplicate of #218 

We may first need to properly detect the Rasberry Pi 3B+ model not to offer 802.11ac  as an option for the rasberry pi's that don't have 802.11ac hardware support.

Juln75, 2018-10-17:
Hi @D9ping ,

Ok, thx !
Buying a wireless usb adapter like this one can be a workaround to use 802.11ac ? ([TP-Link Archer-1300Mbps ](http://a.co/d/dPg1kaU))
Or I will have the same issue ?
Drivers for TL-WN725N V2 - 3.6.11+ -> 4.xx.xx : https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=62371

D9ping, 2018-10-17:
There is no need to buy a wifi adapter as that would leave you with the same problem: It's not configured in hostapd.

Currently, selecting the 5 Ghz wifi channel is not implemented in the RaspAP webinterface and enabling 802.11ac is not even started. Often it's required to use at least Dynamic Frequency Selection(DFS) (ieee802.11h and ieee802.11d) for it being allowed.

ashwin-phadke, 2018-11-14:
` dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
   Loaded: loaded (/lib/systemd/system/dnsmasq.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2018-11-14 19:07:31 IST; 6min ago
  Process: 481 ExecStartPost=/etc/init.d/dnsmasq systemd-start-resolvconf (code=exited, status=0/SUCCESS)
  Process: 388 ExecStart=/etc/init.d/dnsmasq systemd-exec (code=exited, status=0/SUCCESS)
  Process: 360 ExecStartPre=/usr/sbin/dnsmasq --test (code=exited, status=0/SUCCESS)
 Main PID: 480 (dnsmasq)
   CGroup: /system.slice/dnsmasq.service
           └─480 /usr/sbin/dnsmasq -x /run/dnsmasq/dnsmasq.pid -u dnsmasq -r /run/dnsmasq/resolv.conf -7 /etc/dnsmasq.d,.dpkg-dist,.dpkg-old,.dpkg-new --local-service --trust-anchor=.,19036,8,2,49aac11d7b6f6446702e54a1607371607a1a41855200fd2ce1cdde32f24e8fb5

Nov 14 19:12:31 raspberrypi dnsmasq-dhcp[480]: DHCP packet received on wlan0 which has no address
Nov 14 19:12:38 raspberrypi dnsmasq-dhcp[480]: DHCP packet received on wlan0 which has no address
Nov 14 19:12:43 raspberrypi dnsmasq-dhcp[480]: DHCP packet received on wlan0 which has no address
Nov 14 19:12:48 raspberrypi dnsmasq-dhcp[480]: DHCP packet received on wlan0 which has no address
Nov 14 19:12:56 raspberrypi dnsmasq-dhcp[480]: DHCP packet received on wlan0 which has no address
Nov 14 19:13:11 raspberrypi dnsmasq-dhcp[480]: DHCP packet received on wlan0 which has no address
Nov 14 19:13:15 raspberrypi dnsmasq-dhcp[480]: DHCP packet received on wlan0 which has no address
Nov 14 19:13:20 raspberrypi dnsmasq-dhcp[480]: DHCP packet received on wlan0 which has no address
Nov 14 19:13:25 raspberrypi dnsmasq-dhcp[480]: DHCP packet received on wlan0 which has no address
Nov 14 19:13:33 raspberrypi dnsmasq-dhcp[480]: DHCP packet received on wlan0 which has no address
`

This is my `sudo service dnsmasq status`, I am also not able to access the admin URL. I am using ppp0 as my data connection source and not Ethernet. Does this change anything?

rendia, 2019-03-03:
Hi, is there a workaround to use RPI 3B+ as a 5Ghz AC Hotsspot with RaspAP?

closed 2019-03-04 by billz

rendia, 2019-03-05:
Sorry, you just closed it?

And what is the conclusion? There ist no useful answer to find by google.
It would Beste Grüße nice to get an advice. 
Thanx

D9ping, 2019-03-05:
You could enable 802.11ac in HostAPD configuration manually to get 802.11ac wifi. But you are also required to select a proper 5Ghz channel, enable DFS and set hw_mode to a.
A example of all options of how hostapd can be configured can found [here](https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf)