RaspAP/raspap-webgui #255: No internet on wifi ap when using OpenVpn client

bnilesh, 2018-10-20:
Hi thanks for such an awesome utility!

I have used quick setup install script to get my WiFi AP up and running on RPI3. Made suggested changes regarding ipv4 forwarding and iptables updates for getting internet on WiFi AP.
Everything works well when I am not running VPN client. I can connect to WiFi AP and get internet on guest devices. But when I start VPN client on RPI3, I can still connect to WiFi AP but no Internet on guest devices. While I do have internet on RPI3 when using VPN client.

I'm not expert but can provide any additional details/data if needed.

Thanks!

### Your environment
* Raspberry Pi hardware - Pi 3 Model B
* Raspbian Jessie
* RaspAP Quick Install
* Other software or services running with RaspAP - OpenVpn Client

### Steps to reproduce
starting VPN client on RPI3 causes internet disruption on AP connected guest devices.

### Expected behaviour
While RPI3 is running VPN client, guest devices connected on AP should be able to access internet.

### Actual behaviour
No Internet on guest devices connected on AP when RPI3 is running VPN client.

cyberporkexception, 2018-10-29:
Hello,

I have the same issue after adding a wireless connection via the Wifi Client GUI on 10.3.141.1 and a reboot I'm connected to my WLAN but I can't browse in the world wide web... 

Is there a solution for it yet?

Greetings from Austria :)

billz, 2019-04-13:
@bnilesh thanks for the kind words. I'm not able to troubleshoot OpenVPN configs with this project, sorry.

@cyberporkexception I've pushed an update that allows you to host an AP while simultaneously connected as a WiFi client.

closed 2019-04-13 by billz