RaspAP/raspap-webgui #261: Enhancements and bug fix for WiFi Configuration page

PRO2XY, 2018-11-11:
### Enhancements and bug fix for WiFi Configuration page
I have made some changes to [`configure_client.php`](https://github.com/billz/raspap-webgui/blob/master/includes/configure_client.php) and `/etc/sudoers` to:
* Add RSSI / Signal strength field for scanned networks
* Add connect button to connect to a particular SSID
* Make password masked and add 'show' button
* Repair WiFi scanning (add -i option on wpa_cli calls)
* Rename $scanned_networks to $networks as expected in the rest of the document

Files are uploaded in a gist:  [https://gist.github.com/PRO2XY/2ceaeb2d80e32356ea65b6ad60d87a02](https://gist.github.com/PRO2XY/2ceaeb2d80e32356ea65b6ad60d87a02)

I had not git pulled the files initially so I can't create a PR, but maybe someone else can. 

### Expected behaviour
You should be able to see a page like this

[image]

billz, 2018-11-13:
@PRO2XY many thanks. Masking the passphrase has been on my todo list. The RSSI/signal strength and UI enhancements are nice additions as well. I'll test these changes, incorporate them into an update and credit you as the contributor.

billz, 2018-11-13:
Refactored configure client with these enhancements and @njkeng's mobile friendly UI treatment. Looking good and (finally) more manageable on small mobile devices. The passphrase visibility toggle enhances security by discouraging shoulder surfing.
 
[image]

commit d2fc5e42 (2018-11-15): Several enhancements, thanks @PRO2XY & @njkeng. Resolves #168, Resolves #261
closed 2018-11-15 by billz