RaspAP/raspap-webgui #262: Can't Access AP or set Static IP on Armbian Stretch

danielwilson654, 2018-11-15:
### Subject of the issue
dhcpd.conf has no effect in Armbian Stretch

### Your environment
* Armbian Stretch
* RaspAP Quick Install

### Steps to reproduce
Vanilla Arbian Stretch install - Auto install RaspAP - Try access Hotspot or set static IP

### Expected behaviour
Static IP address updates in RaspAP Gui should change set ip address ifconfig

### Actual behaviour
When setting static IP in a RaspAP it reports as applied settings but ifconfig shows the ip address is not updated. I noticed `systemctl status dhcpd` returned `Unit dhcpd.service could not be found.` so I ran `sudo apt install isc-dhcp-server`. Unfortuanly it looks like this service is not compatible with the dhcpd.conf files RaspAp creates.

billz, 2018-11-15:
Armbian Stretch is not supported by this project.

closed 2018-11-15 by billz

danielwilson654, 2018-11-15:
Maybe oneday?