RaspAP/raspap-webgui #266: Using WLAN1 as AP and WLAN2 as Client

SKazukiX, 2018-12-20:
Hello, i have 2 USB Wifi Adapter (WLAN1 & WLAN2) 
I want to make WLAN1 as AP and WLAN2 as Client

Raspberry Pi hardware: Pi 3 Model B
Raspbian version: Stretch
RaspAP

The good thing is that I've succesfully change WLAN1 as AP but WLAN0 is defaulted as Client and Not WLAN2. In Raspi both WLAN0 and 2 are connected to the Wifi but only WLAN1 wont give connection to the AP. Is there any config that i have to change to set WLAN2 as Client ?

enseitankado, 2019-01-30:
You can exlude interfaces from controlling of dhcpcd.
Because of dhcpd forces interfaces to managed mode (interfaces as a wireless client).
Please use denyinterfaces directive in /etc/dhcpcd.conf suc as:

denyinterfaces wlan1

Good luck.

Kzitold, 2019-02-15:
Does this work to daisy chain RaspAP?

Like, if someone were to set up a RaspAP to a VPN-ed desktop on the second floor, and then "this" in the middle for... a Nintendo Switch in the basement?

closed 2019-03-04 by billz