RaspAP/raspap-webgui #267: Reserve IP adress for some mac adress

st4chuu, 2018-12-21:
### Subject of the issue
Try to connect ip address to mac address of clients. I've got device, which can not save static ip address. I would like to reserve some ip adress for some pool of devices, and dhcp will be always give some ip adress for mac adress.

### Your environment
* Raspberry Pi hardware (examples: Pi 3 Model B, Pi 2 Model B, Pi 1 Model B+, Pi Zero)  
Pi 3 Model B
* Raspbian version (examples: Stretch, Jessie)
Stretch
* RaspAP Quick Install or Manual setup?
 Quick Install
* Other software or services running with RaspAP?
nope

### Expected behaviour
As in Subject of the issue. Is it possible?

billz, 2019-01-03:
It's possible, but not implemented.

DD-wrt firmware has a straightforward UI for managing DHCP static leases. Something like this could potentially be added to RaspAP in the future.

[image]

You can reserve static IP addresses for various clients in your network by writing the reservations in `/etc/dnsmasq.conf`. The static lease takes the form `dhcp-host=<mac address>,<ip address>`.

Restart the dnsmasq service and reconnect your reserved clients.

camilomarino, 2019-04-06:
It works perfectly. It would be nice if this was added in the webgui.

PR #361 (2019-08-02): static dhcp leases
this patch adds ui to manage static leases in the dnsmasq config. resolves #267.

[image]

known issues:
- mac and ip addresses don't get validated
- statically assigned ip addesses should always stay outside the dhcp range (i think dhcp servers still don't exclude static addresses from their range?)
closed 2019-08-02 by billz