RaspAP/raspap-webgui #268: NOT Working with pi Zero W (AP+STA)active!

fu-techiot, 2018-12-24:
### Expected behaviour
Tell us what should happen.

### Actual behaviour
Tell us what happens instead.

**Hello There** @billz 
                                I have raspberry pi zero W, i have successfully installed raspAp using QUICK SETUP but the generated AP using another method is gone.

i have one question your method is changing the wifi configuration that are previously made? or all the configuration that previously made and changed it to 

> 
>     IP address: 10.3.141.1
>         Username: admin
>         Password: secret
>     DHCP range: 10.3.141.50 to 10.3.141.255
>     SSID: raspi-webgui
>     Password: ChangeMe
> 

and after reboot it creates AP named raspi-webgui as you mentioned in readme file with above configuration?

I'm using
Raspberry Pi Zero W rev 1.1 
Linux raspberrypi 4.14.79 (stretch)

I had already made Pi  Zero to work in AP+STA  using following simple script 
[https://github.com/lukicdarkoo/rpi-wifi/blob/master/configure](url)
 Or 
[https://blog.thewalr.us/2017/09/26/raspberry-pi-zero-w-simultaneous-ap-and-managed-mode-wifi/](url)

linusaurus, 2019-01-22:
I have been trying to get raspAP to work on zero for weeks, and the links you give are no longer valid. How did you get raspAP to work on a pi since it need wlan0 configured first to get the AP to intall?

billz, 2019-01-22:
@linusaurus have you checked our FAQ? There is a question that deals with this specifically. 
https://github.com/billz/raspap-webgui/wiki/FAQs#pizero

sprokkie, 2019-01-26:
i have the same problem. i lookad at the faq, about the pi zero.
Do i have to do all the steps written down there? ( ifollowd the link)
do i have to do that before installen raspap or after installing ?

is the rasap gui working after the steps ? i love the gui.

billz, 2019-03-04:
This may be helpful https://blog.thewalr.us/2017/09/26/raspberry-pi-zero-w-simultaneous-ap-and-managed-mode-wifi/

Will also update the FAQ accordingly.

closed 2019-03-04 by billz

billz, 2019-12-16:
Managed mode AP is now [a core feature of the project](https://github.com/billz/raspap-webgui#simultaneous-ap-and-wifi-client). It was designed and tested specifically for the Pi Zero W, but will work on any RPi with onboard wireless.