RaspAP/raspap-webgui #269: Feature Request: Support for Debian 9

danielwilson654, 2018-12-29:
This project is the best of it's kind. I think it even has commercial potential.

**Why support Debain?**
This would open up the doors for developers (like myself) to use and adapt this software for more commercial boards that use the rock solid Debian or Armbian distributions. 

The Raspbian community could benefit from the features and stability the commercial developers would be willing to add.

Given that Raspbian is based on Debian, what would adding compatibility involve?

billz, 2019-01-21:
@danielwilson654 thanks for the kind words. 

Without delving too deeply, I spun up Debian 9 (stretch), 64-bit ARM on VirtualBox and had no issues running the RaspAP Quick Installer and accessing the web UI. This virtual environment doesn't have a connected wifi device that supports AP mode, however, so that was the extent of my testing.

Given that it passed this basic compatibility test, I'd suggest trying the installer on "real" (ie, non-virtual) hardware. I don't have any commercial boards handy... just a box of Pi's 😉Any takers?

danielwilson654, 2019-01-26:
@billz 
I'll give it a shot. I'll keep you posted.

danielwilson654, 2019-01-26:
For testing I will use Armbian which is basically Debian GNU/Linux 9 with some compatibility tweaks for the specific hardware. You can read more about it here.
https://www.armbian.com/

danielwilson654, 2019-02-03:
Results
=======

| Item                  | Works | Comment                                                                                                                                                                                  |
|-----------------------|-------|------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------|
| Dashboard             | Yes   | I think                                                                                                                                                                                  |
| Configure WiFi client | Half  | Shows only some wifi networks Worked after I ran armbian-config > Network > Clear blocked networks                                                                                       |
| Configure hotspot     | No    | “HostAPD is Running" Settings can be saved, wifi hotspot is visible but DHCP doesn’t work so connecting to the network fails. It is possible to connect to the network with a static IP. |
| Configure networking  | no    | “Settings successfully applied”  Settings are apparently saved but they have no effect.                                                                                                  |
| Configure DHCP Server | Maybe | “Dnsmasq is running”  Looks all good but it’s obviously not working.                                                                                                                     |
| Configure Auth        | yes   |                                                                                                                                                                                          |
| Change Theme          | yes   |                                                                                                                                                                                          |
| Data usage            | No    | Couldn't see any data coming through                                                                                                                                                     |
| System                | yes   |                                                                                                                                                                                          |

### How to replicate

### System

ARMBIAN 5.65 stable Debian GNU/Linux 9 (stretch) 4.14.78-sunxi

**ifconfig**

enx00e04c534458: flags=4099&lt;UP,BROADCAST,MULTICAST&gt; mtu 1500

 ether 00:e0:4c:53:44:58 txqueuelen 1000 (Ethernet)

 RX packets 0 bytes 0 (0.0 B)

 RX errors 0 dropped 0 overruns 0 frame 0

 TX packets 0 bytes 0 (0.0 B)

 TX errors 0 dropped 0 overruns 0 carrier 0 collisions 0

lo: flags=73&lt;UP,LOOPBACK,RUNNING&gt; mtu 65536

 inet 127.0.0.1 netmask 255.0.0.0

 inet6 ::1 prefixlen 128 scopeid 0x10&lt;host&gt;

 loop txqueuelen 1000 (Local Loopback)

 RX packets 0 bytes 0 (0.0 B)

 RX errors 0 dropped 0 overruns 0 frame 0

 TX packets 0 bytes 0 (0.0 B)

 TX errors 0 dropped 0 overruns 0 carrier 0 collisions 0

wlan0: flags=4099&lt;UP,BROADCAST,MULTICAST&gt; mtu 1500

 ether cc:b8:a8:b4:33:fc txqueuelen 1000 (Ethernet)

 RX packets 0 bytes 0 (0.0 B)

 RX errors 0 dropped 0 overruns 0 frame 0

 TX packets 0 bytes 0 (0.0 B)

 TX errors 0 dropped 0 overruns 0 carrier 0 collisions 0

### Install Armbian

1.  downlaod Armbian(Debian server image from Armbian)
2.  Extract 7zip file
3.  connect to single board computer (SBC) using FEL mode
4.  Clone Arbian .img file to eMMC
5.  Reboot SBC

### Connect SBC to Internet using Ethernet (DO NOT SKIP!)

### Turn on wifi hotspot for easier debugging and auto install software 

1.  Run “sudo armbian-config” go to &gt; Network &gt; Hotspot (this option won’t be visible if you are connected to a wifi network)
    <https://www.youtube.com/watch?v=__BJtInOQFY>
2.  Restart SBC
3.  Find password by running “cat /etc/hostapd.conf” and look for “wpa\_passphrase=”
4.  Test netowrk

### Install RaspAP

1.  Sudo apt-get update
2.  wget -q https://git.io/voEUQ -O /tmp/raspap && bash /tmp/raspap

###

danielwilson654, 2019-02-03:
It looks like Raspbian uses different network configurations from debien under the hood...
https://raspberrypi.stackexchange.com/questions/39785/differences-between-etc-dhcpcd-conf-and-etc-network-interfaces

danielwilson654, 2019-02-03:
If you don't have a SBC compatible with Armbian(or debian) you can run debian on VirtualBox and if you have a USB wi-fi dongle laying around you can attach that to the VirtualBox for testing.

MarkoBursic, 2019-10-17:
Any news? Have you tried with Armbian Buster? It would be a very nice feature having the RaspAP running on RockChip.

billz, 2019-10-20:
@MarkoBursic agreed. I'd really like to add Armbian support for the Rock & Orange Pis. In a middle of #413 at the moment. Will circle back to this soon.

billz, 2020-01-20:
I ordered an Orange Pi 3 + Pi Zero LTS for testing Debian and Armbian installs.

billz, 2020-01-20:
Anecdotally, I've followed forum posts of users successfully running RaspAP on the Orange Pi with Armbian. Will see if I can replicate this.

billz, 2020-03-13:
With some minor changes RaspAP installs and runs on Armbian Buster :) 

Hardware: Orange Pi 3
Kernel: `Linux orangepi3 5.4.20-sunxi64 #20.02.1 SMP Mon Feb 17 02:37:37 CET 2020 aarch64 GNU/Linux`

Follow the project prerequisites per usual. Two updates are required for the installer: 1) change php7.1-cgi to php7.3-cgi, and 2) install dhcpcd5. These are on the `armbian-buster` branch.

Currently running a stable 5GHz 802.11ac AP with several connected clients.
That's enough for tonight

[image]

danielwilson654, 2020-03-14:
Seriously awesome. Is RaspAP ever intended to allow users to switch between
AP mode and client mode purely from the ui?

Daniel

billz, 2020-03-14:
The UI can be used to configure a wireless client, create an AP or both (managed mode AP).

billz, 2020-03-19:
Working on bringing the installer in line with Debian. It's mostly there, save for references to the (now deprecated) `/etc/rc.local`. Related to #127

billz, 2020-03-29:
Resolved with #537.

closed 2020-03-29 by billz

danielwilson654, 2020-03-29:
🥳🥳🥳