RaspAP/raspap-webgui #273: Pi stops being able to find networks on wifi channel 13

Denness, 2019-01-16:
### Subject of the issue
After installing RaspPi the Pi is no longer able to scan and find networks that are on wifi channel 13.

### Your environment
* Raspberry Pi hardware: Raspberry Pi Zero Wireless
* Raspbian version: Raspbian GNU/Linux 9 (stretch)
* RaspAP Quick Install or Manual setup: Quick Install
* Fresh install of OS

### Steps to reproduce
Use the configure client to scan for wifi networks while having a wifi network on channel 13.
or while using
iwlist wlan0 scan

### Expected behaviour
All wifi networks that were found before installing RaspAp should still be found.

### Actual behaviour
Only wifi networks on channels 1-11 are found when scanning

billz, 2019-01-16:
RaspAP filters channel output based on [RF emission regulations for various countries](https://en.wikipedia.org/wiki/List_of_WLAN_channels#2.4_GHz_(802.11b/g/n/ax)). In North America, for example, channel 13 is generally restricted. This filtering only happens in the UI, however. The Quick Installer does not modify the output of `iwlist`.

Have you tried `sudo iwlist wlan0 scan`?

Denness, 2019-01-16:
I'll need to double check that one tomorrow. I may have been mistaken as I was also looking for channels over 13 originally which I later realised were never picked up in the first place

Denness, 2019-01-17:
I doubled checked and I can see all channel 13s on 'iwlist wlan0 scan' and I can also see all networks on channel 13 on the desktop wifi toolbar button which I was sure weren't there yesterday. I am able to find a printer on channel 13 but unable to find the channel I am trying to access on channel 13

Denness, 2019-01-17:
I rebooted the Pi and I am now able to find the network I'm after on channel 13 but I cannot find the printer that was there before. 

This means initial issues is solved but there may be a different issue that all networks are not always found on the RaspAP UI. As I am sure networks that were not found yesterday are found today. 

I'm in a serviced office building which has over 20 networks.

billz, 2019-01-17:
You can compare these results directly via the shell with:

`sudo wpa_cli -i wlan0 scan; sleep 3; sudo wpa_cli -i wlan0 scan_results`

...which is the same method RaspAP uses to populate the client list in the UI. Also bear in mind that the scan results you see in RaspAP will depend on the `country_code` you defined in `hostapd.conf`.

In practice I've found that weak signals will sometimes reappear when you trigger an update with the 'Rescan' button in the UI.

closed 2019-01-18 by Denness