RaspAP/raspap-webgui #274: Pi 3 B+ not showing any network on first try

dalmago, 2019-01-17:
When using a Raspberry Pi 3 B+, the first scan doesn't return any network. I could get it to work by increasing [this sleep time](https://github.com/billz/raspap-webgui/blob/67811c078baad9a6a05438cb6238c32e350c44a8/includes/configure_client.php#L131) to 5 seconds.

Consecutive reloads work as expected.

* Raspbian Stretch
* RaspAP manual install

billz, 2019-01-18:
Thanks @dalmago for sharing your findings. I haven't observed this on the Pi 3B+ with Stretch, but there could be other factors at play. One idea would be to define sleep_time as a constant so it can be easily changed. 

Does the following executed from your Pi's shell return results on the first scan?

`sudo wpa_cli -i wlan0 scan; sleep 3; sudo wpa_cli -i wlan0 scan_results`

dalmago, 2019-01-18:
I've run this before and it returned an empty list, but I just tried and it works fine. 
May it depend on the signal strength (I've changed my board's location)?

billz, 2019-02-07:
Closing this for now. Feel free to reopen if needed. Thanks

closed 2019-02-07 by billz