RaspAP/raspap-webgui #275: Isolation

scargill, 2019-01-22:
Sorry if this has been discussed - this works a TREAT but has full access to the underlying network - I need it to be a guest network which can access the web but not internal resources on the main ip rainge - right now it CAN access the lot. Can anyone help?

Raspberry Pi 3 - on Wheezy - main network is 192.168.14.x - the Pi setup here is as instructed.
Pete

### Before submitting an issue
The [FAQs](https://github.com/billz/raspap-webgui/wiki/FAQs) have answers to many common issues. Please check this before creating a new issue. 

### Subject of the issue
Describe your issue here.

### Your environment
* Raspberry Pi hardware (examples: Pi 3 Model B, Pi 2 Model B, Pi 1 Model B+, Pi Zero)  
* Raspbian version (examples: Stretch, Jessie)
* RaspAP Quick Install or Manual setup?
* Other software or services running with RaspAP?

### Steps to reproduce
Tell us how to reproduce this issue.

### Expected behaviour
Tell us what should happen.

### Actual behaviour
Tell us what happens instead.

billz, 2019-02-07:
It's fairly straightforward to do this with `iptables`. Wheezy is quite an old distribution, though, so I'd strongly recommend you start with RaspAP on a clean install of Stretch Lite. 

I'd also suggest you try asking on  https://raspberrypi.stackexchange.com/ and the related Unix forums. [This thread](https://unix.stackexchange.com/questions/46104/how-to-provide-a-guest-lan-on-one-ethernet-device) should point you in the right direction.

closed 2019-02-07 by billz

MichlW, 2022-12-12:
I managed it using these four rules in iptables:
```
sudo iptables -I FORWARD -i wlan0 -d 10.0.0.0/8 -m state --state NEW -j DROP
sudo iptables -I FORWARD -i wlan0 -d 172.16.0.0/12 -m state --state NEW -j DROP
sudo iptables -I FORWARD -i wlan0 -d 192.168.0.0/16 -m state --state NEW -j DROP
sudo iptables -I FORWARD -i wlan0 -d 172.16.2.254/32 -j ACCEPT
```
First three rules drop any NEW connection from wlan0 to the private ip ranges specified.
Last rule overrides that blocking to accept connections to main net router (which is also my DNS server).
Any Connection from main net to isolated RaspAP net and the clients behind works normally.

I'm not an iptables expert, use it at your own risk. And please let me know if there are any problems with that.