RaspAP/raspap-webgui #277: Raspberry Pi 3B+ setup help

YeshYyyK, 2019-01-26:
### Subject of the issue
I setup my Pi from scratch(and installed NextCloud), then proceeded to use the quick installer script. What do I do next? There is no access point that I can see

### Your environment
* Raspberry Pi 3
* Raspbian Stretch(desktop)
* RaspAP Quick Install
* Nextcloud script(nothing after that, it is not setup completely)

### Steps to reproduce
Tell us how to reproduce this issue.

### Expected behaviour
I should see a access point right? And then be able to get to the web GUI?

### Actual behaviour
There is no access point created, so I'm not able to really do anything

Thaiph1308, 2019-02-21:
> Yes, I have
> […](#)

Did you fixed your isssues?

YeshYyyK, 2019-02-21:
No, I was unable to get it working

billz, 2019-03-04:
Please follow the steps exactly in https://github.com/billz/raspap-webgui/issues/287#issuecomment-469083944

Closing this for now. Reopen if needed, thanks.

closed 2019-03-04 by billz