RaspAP/raspap-webgui #280: Possibility to disable authentication for the web interface? Want to integrate it with FreedomEV project.

jnuyens, 2019-01-27:
### Subject of the issue
Whats the best way to disable authentication for browsers which don't properly support it (and where the web server is only locally accessible)?
Because I would like to integrate it into the FreedomEV project; to configure 'HotSpot Mode' inside an EV like a Tesla.

### Your environment
* Tesla CID running ubuntu 18.04 in a chrooted environment on a NVIDIA Tegra embedded Linux system with touchscreen
* Running into a QTWeb 4 based browser.
* RaspAP Quick Install or Manual setup? Manual setup
* Other software or services running with RaspAP? Nginx with fpm php 7.2

### Steps to reproduce
Get a Tesla, root it and install FreedomEV on a USB stick: http://github.com/freedomev

### Expected behaviour
The Tesla QTWeb based browser seems not to be able to handle authentication, it does not ask for username/password (where it does ask for that when I test with the links browser). It would be nice to be able to just disable it easily.

### Actual behaviour
Nothing visible.

billz, 2019-01-27:
I can safely say this is the first issue re: compatibility with a rooted Tesla 😉

Disabling authentication is generally not recommended, but probably the built-in security of the Tesla is more effective than basic browser auth.

In any case, you can do this by removing or commenting out this line in `index.php`:
`include_once( 'includes/authenticate.php' );`

Would be nice to see a screenshot of RaspAP in the Tesla display if this works for you.

jnuyens, 2019-01-27:
Thanks for the swift reply. Will do, as soon as that works. Are you coming to FOSDEM next weekend? Then I can buy you a beer :-D

jnuyens, 2019-02-01:
[image]
Fun, fun, fun!! :-D

billz, 2019-02-01:
@jnuyens very cool! thanks for sharing!  
I won't make it to FOSDEM this weekend, unfortunately. Hope we can connect for a beer before next year's conference. Enjoy and try to keep your eyes on the road 😅

billz, 2019-11-01:
Marking this as resolved. @jnuyens thanks again for sharing. Best wishes with the Freedom EV project

closed 2019-11-01 by billz

bestkid1234, 2021-03-02:
Hi! I commented the line you said above but the site still pop-up a login dialog. How can i disable this too?

MichlW, 2022-12-10:
I also wanted to disable authentication for web interface but the above solution does not work anymore. So I managed this by editing the file `includes/authencate.php`
from
`$validated = ($user == $config['admin_user']) && password_verify($pass, $config['admin_pass']);`
to
`$validated = true;`

@billz: May this cause any functional problems? I dont want to discuss security aspects on this.

I wonder why I couldnt find an official way to do this.
Maybe my solution works for others as well.