RaspAP/raspap-webgui #283: More detailed instructions on setting this up?

YeshYyyK, 2019-02-02:
Pardon me

How exactly do I setup my Pi 3 to be a WiFi extender(using eth0/Ethernet as the source and broadcasting over its built in WiFi), using this as a frontend?

Do I follow the instructions here: https://github.com/SurferTim/documentation/blob/6bc583965254fa292a470990c40b145f553f6b34/configuration/wireless/access-point.md#using-the-raspberry-pi-as-an-access-point-to-share-an-internet-connection

Or the initial instructions mentioned in that same post?

@billz

billz, 2019-02-07:
Yes, you can apply @SurferTim's changes after running the Quick Installer. Thanks

closed 2019-02-07 by billz

YeshYyyK, 2019-02-07:
Which one? The bottom one or the upper one? Or...both?