RaspAP/raspap-webgui #284: VPN

Kzitold, 2019-02-02:
Pi getting internet from a Windows 10 desktop running NordVPN.
Shared connection from NordVPN instead of Wi-Fi adapter so the Pi goes through my desktops VPN.
It works, but only when my phone is also running NordVPN.

How to make my phone not need NordVPN to connect to Pi Wi-Fi, while also encrypting all data from my Pi? Should I have the Pi running VPN and to Wi-Fi, instead of sending it through my desktops VPN?

--------------------------------
Raspbian Stretch with desktop and recommended software(.zip)
Release date: 2018-11-13

Pi 3

Installed with `wget -q https://git.io/voEUQ -O /tmp/raspap && bash /tmp/raspap`
1.4.0

[My Win10 desktop](https://www.dropbox.com/s/u42sv8zi180jtqa/pcreceipt.png?dl=0)

billz, 2019-02-15:
In this case, you will be better off configuring the RPi with a VPN service. The PiVPN project offers a nice solution with a quick installer, similar to RaspAP http://www.pivpn.io/

closed 2019-02-15 by billz

Kzitold, 2019-02-15:
Oops. Forgot to close this.
It seemed to have fixed itself once I switched to another server on my desktop.

NordVPN #2800