RaspAP/raspap-webgui #290: Question about line 134-136 in configure_client.php

sezhu, 2019-02-13:
### Subject of the issue
I can see all my neighbors' WiFi listed (8 SSIDs) in the client configuration list, but I cannot find my own house's SSID, which has the strongest signal.
I commented out line 134-136 in configure_client.php, then my own house's SSID appeared on the client configuration list.

### Your environment
Pi 3
Raspbian version: Stretch
RaspAP Quick Install
Other software or services running with RaspAP? 
I following the link How-To: Setting up a Raspberry Pi as an access point in a standalone network (Tested with Raspbian Stretch) on your main page before installation of RaspAP. I believe that installed dnsmasq and hostapd daemons.

### Steps to reproduce
If I didn't comment out line 134-136 in configure_client.php, it will cause the problem.

  for( $shift = 0; $shift < 2; $shift++ ) {
    array_shift($scan_return);
  }

### Expected behaviour
I don't know why you want to remove the first 2 clients in the above code, one of which clients is my own house's SSID.

### Actual behaviour
My own house's SSID disappeared from the client configuration list.

MiloMindbender, 2019-02-15:
I noticed a similar problem, at my office there are over 20 APs visible to my phone but on the Raspap client page only about half of them were showing up.  All of them seemed to be showing up in the raspberry pi's desktop WiFi widget.

Unfortunately the fix required to get the AP working (see https://github.com/billz/raspap-webgui/issues/287) kills wpa_supplicant, after that none of the clients show up.

sezhu, 2019-02-18:
I figured out what the problem is.

Line 134-136 is to remove the 2 title lines from wpa_cli results.

However, my wpa_cli (ver 2.4) results only consist 1 title line in the following:
bssid / frequency / signal level / flags / ssid

So my 1st SSID result line following the title line will  be removed by this mistake.

My solution is simple, just comment line 134, and line 136 and keep line 135 will fix the situation.

sezhu, 2019-02-18:
MiloMindblender:

Have you tried 
wpa_cli reconfigure

That might work. I think the wpa_supplicant needs to be reconfigured if it was messed up.

billz, 2019-03-01:
This was originally reported in #188 and fixed. It looks like the bug was reintroduced when the configure_client page was refactored. I'll push the fix (again). Thanks

commit 6a01eed1 (2019-03-01): (re)fix of #188. Resolves #290
commit 543a5e09 (2024-05-11): Add net-tools dependency to Armbian, resolves #290
closed 2019-03-01 by billz