RaspAP/raspap-webgui #291: Cant change b/g/n

crazykakaroto, 2019-02-14:
### Subject of the issue
I Can't activate the 802.11 n mode, i select in menu, but after conect i see the properties in windows and show as g mode, same thing when conect my phone.
 And in the dashboard dont show the wireless information, only sais that i am not conected ( in the hotspot mode, i dont tested in client mode)

### Your environment
* Raspberry Pi hardware Pi 3 Model B+
* Raspbian version  Stretch
* RaspAP Quick Install 
* Other software or services running with RaspAP? NO

billz, 2019-03-03:
On the **Configure hotspot** > **Security** tab, be sure to select CCMP for the Encryption Type. Save the settings and restart the hotspot. The wireless mode should be reported on clients as 802.11b/g/n.

```
RaspAP:
  PHY Mode:		802.11n
  Channel:		1
  Network Type:		Infrastructure
  Security:		WPA2 Personal
  Signal / Noise:	-49 dBm / -86 dBm
  Transmit Rate:	73
```

If using TKIP for encryption with WPA, you will be restricted to 54 Mb/s. This is because the IEEE 802.11n draft prohibits using high throughput with WEP or TKIP ciphers.

Re: Wireless Information statistics, these will appear when you are connected in client mode.

crazykakaroto, 2019-03-04:
it worked, thank you!!

closed 2019-03-04 by crazykakaroto