RaspAP/raspap-webgui #292: Raspberry Pi 3B+ Strech setup

Thaiph1308, 2019-02-21:
### Subject of the issue
After install RaspAP, i Can see the AP, everything is ok. But After i setting 

### Your environment
* Raspberry Pi B3+
* Raspbian version Stretch
* RaspAP Quick Install 
* 

### Steps to reproduce
Install new Stretch
SSH to Raspberry
Install RaspAP
Reboot RP
Connect to RP AP
Select my home network and enter the Wifi password
### Expected behaviour
Raspberry can connect to my home network and also has an AP mode
After connecting to my home network, the AP signal is lost, I can't see an AP more

sezhu, 2019-02-22:
I think the developer of the project apparently just wanted to create a GUI, and not a real world web site for this project.
So the project would run perfectly if you hook up your Rasp Pi with a monitor. Even the site 10.3.141.1 is accessible after you entered Passphrase, and after you clicked Connect button.
However, if you didn't hook up your Rasp Pi with a monitor, but instead using a phone or PC to connect the raspap-webgui WiFi, it's buggy. It's always a headache to deal with the bug. The raspap-webgui immediately disappeared after you entered Passphrase for your home WiFi.

My solution is simple:
Comment out line 114, 115, 118, 119 & 120 in configure_client.php

After you finished type in Passphrase and click Connect, you need to restart the Rasp Pi.

billz, 2019-03-04:
Duplicate of #272 #271 #200

closed 2019-03-04 by billz