RaspAP/raspap-webgui #294: Feature Request: Allow Installation into existing Web Server

ZacAttack24756, 2019-02-27:
It would be a setting in the autoinstallation that asks if I would want to install it to a WebServer already installed onto my Pi, like Lighttpd, or Apache. It would then ask for a subdirectory of the `/var/www/html` folder. This is useful especially for me because I use my Pi as a server for other applications and a Hotspot / Access Point.

ZacAttack24756, 2019-02-27:
**I don't really know PHP or any related languages

billz, 2019-11-05:
@ZacAttack24756 thanks for the suggestion. This has now been implemented

closed 2019-11-05 by billz

sedgett, 2019-11-08:
Just wanted to say love this enhancement!   I'm running a stock copy at the /admin URI and a hacked up version at / that shows the status of the uplink  without requiring authentication.

As a aside, in configuring an install that has everything but the dashboard disabled I noticed that the "System" table doesn't have a config option to disable.   Also, in monitor mode, webconsole appears to still be active.  

Thanks for this!

Thatoo, 2020-01-01:
There is apparently a way to change lighttpd web server port explained in the FAQ : https://github.com/billz/raspap-webgui/wiki/FAQs#webport

commit 25c4b965 (2019-11-05): Added install option for web root. Resolves #294