RaspAP/raspap-webgui #295: Feature Request: Prevent WAN access to Web GUI

FrandSX, 2019-03-01:
Hi there,

I've set up a clean Raspberry Pi 3B+ with a fresh Raspbian Lite.
Installed hostapd and dnsmasq according to
https://github.com/SurferTim/documentation/blob/6bc583965254fa292a470990c40b145f553f6b34/configuration/wireless/access-point.md

I did not set up a bridge device, because I prefer the Pi to function as a DHCP/NAT device. Everything seems to work properly, clients connect to the AP, receive IP addresses, and are able to browse online.

However, upon testing, I found that the RaspAP web portal is accessible from the WAN side. This seems to be a potential security issue.

I was unable to find any option the GUI to disable this behavior. Have I perhaps misconfigured something, or are there simple config options that I have overlooked? One possible workaround I've managed to come across is to set up an iptables rule to prevent port 80 access from WAN.

quamby, 2019-09-14:
Apologies if this is a newbiosh answer.. I don't understand this very well..

You can control this through lighttpd config (found at /etc/l;ighttpd/lighttpd.conf)

By default lighttpd listens on any ipaddress configured on the system and responds to a request on port 80 .

Check this by running `sudo netstat -ntlup | grep :80`   and you'll see a line that says

> tcp  0  0.0.0.0:80     0.0.0.0:*    LISTEN   ###/lighttpd

This means that lighttpd (ie the web server) is listening on any ip address and will respond to any http request.

If you don't want that, then yo need to specify the listening address in the lighttpd.conf file.

Sorry for the formatting etc.. I hope it is useful..

The first block` <b.b.b.b> `sets the ip address lighttpd will listen on --> this is the config you want to change to prevent the web server from responding to requests on the wan side.

the greyed out block below this starting with _$HTTP["scheme"]_  is the generic redirect of all http traffic to the address set in <b.b.b.b>  to https (refer the FAQ for a better  explanation of how to do this up).

The bolded line  is the directory path to where your certificate is stored.

The next line of code sets lighttpd to listen for HTTPS requests to the UI on interface with ip address <a.a.a.a> on port ### (whatever number you choose).

=============================================================================[

server.modules = (
	"mod_access",
	"mod_alias",
	"mod_compress",
 	"mod_redirect",
)

#server.document-root      = "/var/www/html"
server.document-root        = "/var/www/html"
server.upload-dirs             = ( "/var/cache/lighttpd/uploads" )
server.errorlog                  = "/var/log/lighttpd/error.log"
#server.errorlog                = "/usr/src/tech-shield/logs/errors.log"
server.pid-file                    = "/var/run/lighttpd.pid"
server.username              = "www-data"
server.groupname            = "www-data"
```
server.port                      = 80
server.bind		         = "<b.b.b.b>"
```

> $HTTP["scheme"] == "http" {
>     # capture vhost name with regex conditiona -> %0 in redirect pattern
>     # must be the most inner block to the redirect rule
>     $HTTP["host"] =~ ".*" {
>         url.redirect = (".*" => "https://%0$0")
>     }
> }

`$SERVER["socket"] == "<b.b.b.b>:###" {`
ssl.engine = "enable"
**ssl.pemfile = "<<<path/to/your/ssl/pem/file.pem>>>"**
}

```
$SERVER["socket"] == "a.a.a.a:445" {
ssl.engine = "enable"
ssl.pemfile = "<<<path/to/your/ssl/pem/file.pem>>>"
}
```

index-file.names            = ( "index.php", "index.html", "index.lighttpd.html" )
url.access-deny             = ( "~", ".inc" )
static-file.exclude-extensions = ( ".php", ".pl", ".fcgi" )

compress.cache-dir          = "/var/cache/lighttpd/compress/"
compress.filetype           = ( "application/javascript", "text/css", "text/html", "text/plain" )

# default listening port for IPv6 falls back to the IPv4 port
include_shell "/usr/share/lighttpd/use-ipv6.pl " + server.port
include_shell "/usr/share/lighttpd/create-mime.assign.pl"
include_shell "/usr/share/lighttpd/include-conf-enabled.pl"

A script that will configure lighttpd to listen natively on the LAN side and redirect HTTP requests to HTTPS, and then listen on HTTPS only on the WAn side will look something like this:
```
!#/bin/bash

# This will determine what interface your wlan0 ip is
_ip1=$(hostname -I | cut -d " " -f1)

# This will determine what your uap0 address is
_ip2=$(hostname -I | cut -d " " -f2)

# set this to whatever port you want
###=12345

sed -i "s/a.a.a.a/$_ip1/"  /etc/lighttpd/lighttpd.conf
sed -i "s/b.b.b.b/$_ip2/"  /etc/lighttpd/lighttpd.conf
sed -i "s/###/$_ip1/"  /etc/lighttpd/lighttpd.conf
exit 0
```

FearNaBoinne, 2020-01-30:
This works, but ideally it should be a check-box in the UI "Allow WAN access to Web UI". After all, this project is about creating an easy to use Wireless AP solution with the Pi

rugk, 2020-07-27:
IMHO it would be best to expose the "bind configuration" in the web interface. So you can configure there, at which address (and port – but this is something you can already configure) it should listen.

billz, 2020-07-29:
The web UI is protected with an encrypted password, so it's not really accessible on the WAN side as described. However, it's a potential security issue for installs with default passwords.

I think @quamby's suggestion using lighttpd.conf is the best approach here, and I agree with @rugk that giving the user control of the bind address on the **System > Advanced** tab makes sense.

PR #648 (2020-08-06): Feature: Add server.bind address option to System UI
This exposes the web server bind address, making it a configurable option on the **System > Advanced** tab.

[image]

Resolves #295
closed 2020-08-06 by billz