RaspAP/raspap-webgui #298: no wifi ssid on my devices around and a solution

lep0le, 2019-03-07:
### Subject of the issue
after a fresh stretch full install an update/upgrade and run the quick install script NO ssid or access to the AP. but find a soltiion in a post here...

### Your environment
* Raspberry Pi Pi 2 Model B,
* Raspbian version Stretch full
* RaspAP Quick Install 
nope
### Steps to reproduce
reboot and always no ssid.

### Expected behaviour
a mystery wlan0 don't start an host

### Actual behaviour
nothing

BUT a solution: running the following command start the AP
`sudo hostapd -dd /etc/hostapd/hostapd.conf`

so as it start the AP i create a script hostapscript.sh in pi home
with that command inside
`
#!/bin/bash
sudo hostapd -dd /etc/hostapd/hostapd.con
done`
Change the chmod of the script with : 
`sudo chmod +x hostapscript.sh`

then at the end of bashrc
`sudo nano .bashrc
`
 i put this line
`./hostapscript.sh`

now it work...

WHY but why ????

lep0le, 2019-03-07:
hum hum 
my solution need to autostart lxterminal
so `mkdir /home/pi/.config/autostart`
then `nano /home/pi/.config/autostart/AP.desktop`
and paste `[Desktop Entry]
Type=Application
Name=autostart_AP
Exec=lxterminal
`
and that's it

thyristor59, 2019-03-08:
Hi, I had terrible complications just like you.
to pull out the hair ....
your order is finally working !!!
I see the raspi-webgui and connect by VNC without worry!
however, I can not start the automatic command at startup!
I am a novice.
Do you have another way of doing or more details to understand how it works and how to better achieve!
sorry for english i'm french

billz, 2019-03-12:
This was reported elsewhere, see https://github.com/billz/raspap-webgui/issues/287#issuecomment-471038242. A new version of hostapd introduced a bug. I updated the installer to resolve this.

> so as it start the AP i create a script hostapscript.sh in pi home with that command inside

Alternatively, you can either 1) run the install again, which will fix this issue, or 2) execute the following: `sudo systemctl unmask hostapd && sudo systemctl enable hostapd` which achieves the same thing. 

Hostapd should autostart on its own after you've done this. The script won't be necessary.

billz, 2019-03-17:
Closing this for now. Feel free to update if your issue persists, thanks.

closed 2019-03-17 by billz