RaspAP/raspap-webgui #300: Easily switch between wired AP and Wifi client AP mode

billz, 2019-03-12:
This has become a popular requested feature. The use case was captured well by @hallard who also described a procedure for enabling this on the RPi Zero W (later added to our [FAQs](https://github.com/billz/raspap-webgui/wiki/FAQs)). 

I've done some work on a feature branch that enables this with a toggle switch on the **Advanced** tab of **Configure hotspot**. In addition to managing several configuration settings, the order and timing of starting up network services is important to avoid race conditions. 

[image]

So far it's tested well on a Pi 3B+ with Stretch Lite. 

Tracking this with PR #297. 
Related issues: #272 #271 #151 #78, possibly #201

billz, 2019-03-14:
Tested successfully on the Pi Zero W with the Quick Installer.
Updating this issue while connected to RaspAP in WiFi client AP mode 🎉

Dude4Linux, 2019-03-15:
@billz - I'm attempting to use your `raspap-webgui` on an Rpi 3B configured as a `travel router`.  In my use case, eth0 is unused and I want to use an external usb WiFi adapter as `wlan1`.  I need to have `hostapd` running on the built-in adapter, `wlan0`, while the WiFi client uses the external adapter, `wlan1`.  Since I want both modes active simultaneously, I'd rather see a configuration option under `System` or `Configure WiFi client`.

Dude4Linux, 2019-03-16:
@billz - I found an older discussion thread, issue #196, which basically asks for the same thing that I need.  I saw your suggestion there:

> You can change the default setting in /includes/config.php:
> define('RASPI_WIFI_CLIENT_INTERFACE', 'wlan0');define('RASPI_WIFI_CLIENT_INTERFACE', 'wlan0');

I was able to change the WiFi interface to `wlan1` but I had to make the changes to `/etc/sudoers` and figure out how to get `stretch` to stop using `mac` based interface names.

billz, 2019-03-17:
@Dude4Linux indeed, the default wifi interface can be changed in the config. In a future update I'll add the ability to set this via the UI. 

Predictable network interface names can be enabled/disabled with `raspi-config`.

Dude4Linux, 2019-03-19:
@billz - I tried using `raspi-config` to disable predictable network names, but it created a bigger problem. Disabling predictable network names using either `raspi-config` or the grub command line method, results in `wlan0` being assigned to the USB WiFi dongle and `wlan1` being assigned to the internal RPi WiFi which is the opposite of what I need.  I don't want to have to have to modify `raspap-webgui` to use `wlan1` for `hostapd`.  When using `jessie`, I was able to write `udev rules` to override this behavior and assign the interface names based on mac addresses.  This was ugly, but at least it worked.

After upgrading to `stretch`, I find that the `udev rules` no longer work.  Searching for a work-around, I found a post by TheDiveO on [AskUbuntu](https://askubuntu.com/questions/947519/ubuntu-17-04-how-to-rename-a-usb-network-interface-based-on-path-not-based-on).  I tried his suggestion of linking a udev rule
```
sudo ln -s /lib/udev/rules.d/80-net-setup-link.rules /etc/udev/rules.d/80-net-setup-link.rules
```
and magically, it seems to work without needing to add any additional changes (that I'm aware of) and with predictable network names enabled. I'm currently planning to test the method in other situations to make sure that it's the only change required.

Dude4Linux, 2019-03-25:
Well, it turns out that I was wrong about the softlink above fixing the problem.  I was working on an older RPi that had been upgraded from Jessie to Stretch.  Restarting udev after making the change above, reactivated the rules and programs associated with `75-persistent-net-generator.rules` which are part of Jessie but are no longer included in Stretch.

The lines from `73-usb-net-by-mac.rules`
```
ACTION=="add", SUBSYSTEM=="net", SUBSYSTEMS=="usb", NAME=="", \
    ATTR{address}=="?[014589cd]:*", \
    TEST!="/etc/udev/rules.d/80-net-setup-link.rules", \
    TEST!="/etc/systemd/network/99-default.link", \
    IMPORT{builtin}="net_id", NAME="$env{ID_NET_NAME_MAC}"
```
show that the link to `/etc/udev/rules.d/80-net-setup-link.rules` is just another way to disable the predictable network names.

I've tried to come up with a method that will work for both Jessie and Stretch.  Currently, putting the following modified file in `/etc/udev/rules.d/80-net-setup-link.rules` seems to be working for Stretch.
```
# assign onboard devices first by matching RPi MAC prefix
# predictable network names must be disabled if this file is renamed

SUBSYSTEM!="net", GOTO="net_setup_link_end"

IMPORT{builtin}="path_id"

ACTION!="add", GOTO="net_setup_link_end"

IMPORT{builtin}="net_setup_link"

ENV{MATCHADDR}=="b8:27:eB:*", NAME=="", ENV{ID_NET_NAME}!="", NAME="$env{ID_NET_NAME}"

ENV{MATCHADDR}!="b8:27:eB:*", NAME=="", ENV{ID_NET_NAME}!="", NAME="$env{ID_NET_NAME}"

LABEL="net_setup_link_end"
```
This both replaces the original rule, `/lib/udev/rules.d/80-net-setup-link.rules`
and disables the rule, `/lib/udev/rules.d/73-usb-net-by-mac.rules`

TBH, I don't know enough about `udev` to explain why this works better than the original rule unless there is some kind of race condition.

billz, 2019-04-08:
Tested with wired ethernet and as a wireless client ("managed" mode) using RPi 3B & Stretch Lite in 3 different scenarios:

1. Wired AP (eth0). Run Quick Installer & reboot. AP is up and functional with no changes to default config.
2. Wifi client on local network (wlan0). Connect to wireless router as a managed device.
3. Simultaneous AP and managed mode Wifi (uap0). When configured in managed mode, enable AP from Advanced hostapd setting toggle. AP is functional without restart. 

Repeated tests 2 & 3 with RPi Zero W (Stretch Lite) also with Quick Installer. Stopping/starting AP presented no issues. AP in managed mode is observed to be stable on both RPi 3B and Pi Zero W. A sample throughput analysis on the Pi Zero W is below.

[image]

Note: Simultaneous AP and managed mode Wifi uses a known-good default config. Changes to dhcp server settings in this mode are ignored. 

Feature branch merged into master. Please report any issues here. Thanks

billz, 2019-04-13:
Resolves #272 #271 #181 #151 #78 #200 #201

closed 2019-04-13 by billz