RaspAP/raspap-webgui #301: "Hello" instead of the dashboard

BorisKourt, 2019-03-16:
### Subject of the issue
After a successful install on a fresh raspberry Pi the server only responds with "Hello" when going to any address on the default raspberry Pi IP, or `raspberrypi.local`

The network is live and I can connect, but doesn't go up on every reboot.

### Your environment
* Pi Zero W V1.1
* Raspbian GNU/Linux 9 (stretch)
* Quick Install
* Tested with following modules on and off: dwc2, g_ether

### Steps to reproduce
I just did the Quick Install, rebooted and went to the IP

### Expected behavior
Dashboard should appear and prompt for username & password.

### Actual behavior
"Hello" appears instead, no password prompt.

BorisKourt, 2019-03-16:
I've attempted the additional steps presented in the FAQ: https://github.com/ch2i/LoraGW-Setup/blob/master/doc/AccessPoint.md

Unfortunately that appears to have made things works. AP doesn't appear, no longer possible to ssh via USB and dwc2.

billz, 2019-03-17:
@BorisKourt a couple of items here: 1) I have a feature branch in development which adds support for the Pi Zero (see #300), 2) the "hello" response you're seeing is not from RaspAP or the default lighttpd index page. Do you have other web services installed on your Pi?  

Try `netstat -tulpn | grep :80` to confirm which process is listening on port 80. You can also check the contents of `/var/www/html` and see if something other than RaspAP is handling requests.

billz, 2019-04-03:
Closing due to inactivity. Feel free to update with more info

closed 2019-04-03 by billz