RaspAP/raspap-webgui #303: Manual instructions are outdated and no success in working

fireheadman, 2019-03-20:
Attempted the manual installation first (clean build of stretch-lite).
Was unable to use the GUI to set static IPs.

I am running PI3-B model (wlan0) and also using a Panda 300Mbps usb w/extender.
Plan was to use wlan0 to be configured as management and to connect to home network when close by.    Would use wlan1 (Panda dongle) as hotspot network [192.168.3.0 net]
My goal is to use this in my Travel Trailer so that I can host a small network for Plex, media downloads (from cameras) using a small NAS . (everything is PI-based).

Reloaded stretch, this time performed the quick install via cli and I could now update network.  Noticed the sudoers entries were different too.... so unsure what else is happening (without dissecting the install script).

Few hours later.... still unable to get wlan0 or wlan1 to connect a client to the hostspot.
It appears if you select NONE on the security it will not connect... console complains about key not being within correct count, but I told it NONE, No Security, so why would that even matter???    So I put a key in... still no dice.

Also seeing random: Cannot read from /dev/random: Resource temporarily unavailable
in the console....   also see: WPA: Not enough entropy in random pool for secure operations - update keys later when the first station connects

Just one thing after another, so I gave up.
THe GUI is slick!... I'll give you that.  But it does nothing for me if I cannot get it working.

Please catch the documentation up to your installation method(s)/expectations.
I am open for another go if you get it ironed out.

billz, 2019-03-20:
Duplicate of #196

closed 2019-03-20 by billz

billz, 2019-03-20:
Try the Quick Installer with a fresh Stretch Lite image and change `RASPI_WIFI_CLIENT_INTERFACE` to wlan1 in `config.php` after a reboot. Other RaspAP users have reported success with this. I have no experience with the Panda dongle so can't vouch for its compatibility with this project. 

Also, please be aware that the manual instructions are for typical AP setups, and therefore have been condensed to cover this basic use case. The manual steps are not intended to be a tutorial for all possible AP configurations. Public RPi forums are a better bet for support or tutorials.

fireheadman, 2019-03-20:
Thank you I will give it another shot this evening
The panda usb WiFi was plug-n-play on my PI2 model.   Per this document it
shows to be supported on the RBPi3.
https://elinux.org/RPi_USB_Wi-Fi_Adapters

One thing that is not so great is the interface shows up with the non
friendly name and not wlan1.  Not a big deal.  I’ve also read that some
users have needed to update the sudoers file to accommodate the changes.

fireheadman, 2019-03-21:
well.... made it a little further.   Changed the wifi dongle out (panda)
for a TL-WN725Nv2 (which is supported out of the box).   Read the Panda
dongle is prone to dropping connections after 1 hour.
So I have wlan0 = PI3 internal wifi acting as the HotSpot and wlan1 =
TL-WN725Nv2 acting as my live network to outside.

I get the prompt for my HotSpot (using a Macbook Pro) but every time I
insert the password it tells me it is incorrect.
Cannot get passed this part to save my life.

billz, 2019-04-10:
@fireheadman re: password being refused, enable logging and paste the contents here. You can also execute `sudo hostapd /etc/hostapd/hostapd.conf`, try connecting a client to the AP and monitor the output.

fireheadman, 2019-04-12:
Hi Billz, I am abandoning this project in favor of openwrt.
I appreciate you reaching out to me.