RaspAP/raspap-webgui #306: Run RaspAp on ReadOnly FS

matzrm, 2019-03-28:
### Your environment
* Pi 3 Model B+
* Stretch Lite
* RaspAP Quick Install
* Pi Read Only script by Adafruit

### Steps to reproduce
On fresh stretch lite install RaspAp Quick Install and ReadOnly Adafruit script
Then append to /etc/fstab

tmpfs /var/log/lighttpd tmpfs nodev,nosuid 0 0

and change dnsmasq dhcp_lease file position adding in /etc/dnsmasq.conf this :

dhcp-leasefile=/tmp/dnsmasq.leases

### Expected behavior
 I think that would be great have a RASPI_READ_ONLY_FS variable inside config.php to handle readonly filesystem performing a remount rw/ro before/after each shell command.

### Actual behavior
Web-Gui is showed but obviously it cannot saves the changes or reads the network status.

pietrondo, 2020-07-18:
is possible? how can i add a command in the webgui?

ericraymond, 2020-08-02:
Perhaps an easier way to handle this is to allow the user to configure two hooks that gets called before and after commands that need to write to the file system.

In the case of https://gitlab.com/larsfp/rpi-readonly, we would use `rw` and `ro`

This seems better, simpler than detecting it.  The only issue is that if the user has temporarily enabled rw mode, this would revert that.  So maybe a (configurable) test would be nice.

This is far more important than it seems.  I'm tired of having to rebuild a raspap server because the filesystem was corrupted.

mattenz, 2020-10-04:
This functionality would be great for my use case.

I use a Pi Zero W as a dashcam drive for my Tesla using [this project](https://github.com/marcone/teslausb). I was hoping to use RaspAP as a means to easily join the Pi to new wifi networks as I travel so the Pi can connect to my home OpenVPN connection and upload any dashcam footage. The TeslaUSB project makes the SD card read only so that it doesn't get trashed when the car abruptly cuts power to the Pi when it goes to sleep but that obviously conflicts with RaspAP as any changes made in the web gui can't be saved.

Dude4Linux, 2021-03-23:
Log2ram by azlux handles the file logging (https://github.com/azlux/log2ram). I install it by default on all my RPi projects. It saves the log data to RAM and then periodically (daily or weekly) syncs and compresses the logs to disk i.e. flash memory. I don't believe it handles the DHCP leases which is a good point that I hadn't thought of.  Perhaps it could be adapted to include lease files and other types of cached files.

billz, 2021-06-06:
Give https://github.com/RaspAP/raspap-tools a look, in particular `raspian_min_write.sh`. This moves several temp and log files to RAM. Credit to @zbchristian for this repo.

closed 2024-02-08 by billz