RaspAP/raspap-webgui #310: Android phone cannot obtain IP adress

impulse11, 2019-04-10:
Android device cannot obtain an IP adress

Running a Pi 3B+ with stretch lite.
Used quick install.

I changed the network name to "llewellyn2" and psk to "llewellyn".  I also changed the network band to n. Please note I had the same issue before I changed these settings.

I should be able to connect to the hotspot with my android device, but I have issues obtaining an IP address. The interface sometimes changed to "lo" for some reason, so I changed it back to wlan0.

billz, 2019-04-11:
Enable logging and check the output.

impulse11, 2019-04-11:
I enabled it and it didn't log anything

impulse11, 2019-04-11:
Do I need to enable WiFi client?

billz, 2019-04-11:
Enable Logfile output > Save settings > reload Configure hotspot

impulse11, 2019-04-12:
Output:

Configuration file: /etc/hostapd/hostapd.conf
HT (IEEE 802.11n) with WPA/WPA2 requires CCMP/GCMP to be enabled, disabling HT capabilities
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:11:ba:4b and ssid "llewellyn2"
wlan0: interface state COUNTRY_UPDATE->ENABLED
wlan0: AP-ENABLED

impulse11, 2019-04-12:
Output after trying to connect:

Configuration file: /etc/hostapd/hostapd.conf
HT (IEEE 802.11n) with WPA/WPA2 requires CCMP/GCMP to be enabled, disabling HT capabilities
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:11:ba:4b and ssid "llewellyn2"
wlan0: interface state COUNTRY_UPDATE->ENABLED
wlan0: AP-ENABLED 
wlan0: STA f0:d7:aa:31:8b:55 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED f0:d7:aa:31:8b:55
wlan0: STA f0:d7:aa:31:8b:55 RADIUS: starting accounting session A650B0586AD1A6C9
wlan0: STA f0:d7:aa:31:8b:55 WPA: pairwise key handshake completed (WPA)
wlan0: STA f0:d7:aa:31:8b:55 WPA: group key handshake completed (WPA)
wlan0: STA f0:d7:aa:31:8b:55 IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED f0:d7:aa:31:8b:55
wlan0: STA f0:d7:aa:31:8b:55 IEEE 802.11: disassociated
wlan0: STA f0:d7:aa:31:8b:55 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED f0:d7:aa:31:8b:55
wlan0: STA f0:d7:aa:31:8b:55 RADIUS: starting accounting session A97F0812404F5BDB
wlan0: STA f0:d7:aa:31:8b:55 WPA: pairwise key handshake completed (WPA)
wlan0: STA f0:d7:aa:31:8b:55 WPA: group key handshake completed (WPA)
wlan0: STA f0:d7:aa:31:8b:55 IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED f0:d7:aa:31:8b:55
wlan0: STA f0:d7:aa:31:8b:55 IEEE 802.11: disassociated

billz, 2019-04-12:
Select CCMP as the encryption type and restart the hotspot.

impulse11, 2019-04-12:
Logfile after updating encryption and restarting hotspot:

Configuration file: /etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:11:ba:4b and ssid "llewellyn2"
wlan0: interface state COUNTRY_UPDATE->ENABLED
wlan0: AP-ENABLED 
wlan0: STA f0:d7:aa:31:8b:55 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED f0:d7:aa:31:8b:55
wlan0: STA f0:d7:aa:31:8b:55 RADIUS: starting accounting session 1857A8485C95B540
wlan0: STA f0:d7:aa:31:8b:55 WPA: pairwise key handshake completed (RSN)
wlan0: STA f0:d7:aa:31:8b:55 IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED f0:d7:aa:31:8b:55
wlan0: STA f0:d7:aa:31:8b:55 IEEE 802.11: disassociated

impulse11, 2019-04-13:
On the dashboard, under interface (wlan0), it says that an ipv4 address cannot be found. Is this the issue?

impulse11, 2019-04-13:
[image]

billz, 2019-04-13:
> On the dashboard, under interface (wlan0), it says that an ipv4 address cannot be found. Is this the issue?

If using the built-in WiFi for hostapd, yes.
Try restarting wlan0 and capture any error messages with:
```
sudo ifdown wlan0
sudo ifup wlan0
```

impulse11, 2019-04-14:
After running these commands I got these errors:

 [image]

billz, 2019-04-14:
Are you using a USB dongle or the onboard Wifi?
Paste the output of these commands: `iwconfig`, `lsusb`, `lsmod`

impulse11, 2019-04-15:
Onboard WiFi. 

iwconfig:
 [image]

lsusb:
[image]

lsmod:
[image]

billz, 2019-04-15:
Are you connecting to your Pi Zero with USB OTG?

impulse11, 2019-04-15:
I'm using a Pi 3B+ and SSH-ing into it

billz, 2019-04-15:
I'm unable to assist you with troubleshooting this further. Be sure to follow the project prerequisites and try the Quick Installer again. Failing that, use the manual installation steps. Best of luck

closed 2019-04-15 by billz

abhinovpankaj, 2024-12-19:
I am trying on Pi zero 2w , android devices are not able to connect to the hotpot. it just shows obtaining IP address.

billz, 2024-12-19:
Please refer to the FAQ https://docs.raspap.com/faq/#noip. This is not a troubleshooting forum.