RaspAP/raspap-webgui #311: Access point dies after about half an hour

johan-gorter, 2019-04-11:
RPi 3 Model 

### Subject of the issue
I installed a on a fresh Stretch install  on a RPi 3 Model B (not plus). The access point seems to work fine, except after a while (about half an hour maybe) the access point suddenly disappears. Suspicous messages are reported in dmesg. (see below)

### Your environment
* Pi 3 Model B 
* Raspbian version: Stretch DESKTOP
* RaspAP Quick Install
* No other software or services running

### Steps to reproduce
Do a fresh install. Leave the access point unattended for about half an hour.

### Expected behavior
The access point remains for several weeks without any problems.

### Actual behavior
The access point disappears.

Logs:
dmesg:
...
[    5.269703] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    5.279038] brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43430-sdio.bin for chip 0x00a9a6(43430) rev 0x000001
[    5.279310] usbcore: registered new interface driver brcmfmac
[    5.552440] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Oct 23 2017 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f
[    5.553382] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 7.11.15 Compiler: 1.24.2 ClmImport: 1.24.1 Creation: 2014-05-26 10:53:55 Inc Data: 9.10.39 Inc Compiler: 1.29.4 Inc ClmImport: 1.36.3 Creation: 2017-10-23 03:47:14 
...
[ 1260.110929] brcmfmac: brcmf_sdio_hostmail: Unknown mailbox data content: 0x40012
[ 1277.515199] brcmfmac: brcmf_do_escan: error (-110)
[ 1277.515212] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[ 1337.595318] brcmfmac: brcmf_do_escan: error (-110)
[ 1337.595330] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
...
[ 1397.595528] brcmfmac: brcmf_do_escan: error (-110)
[ 1397.595538] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[ 1457.595672] brcmfmac: brcmf_do_escan: error (-110)
[ 1457.595688] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[ 1517.595790] brcmfmac: brcmf_do_escan: error (-110)
[ 1517.595807] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[ 1526.395716] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ 1528.955773] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110

ifconfig (same before and after access point disappears):

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.3.141.1  netmask 255.255.255.0  broadcast 10.3.141.255
        inet6 fe80::1c73:c663:91d8:9d6a  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:e4:6b:f6  txqueuelen 1000  (Ethernet)
        RX packets 124  bytes 27945 (27.2 KiB)
        RX errors 0  dropped 1  overruns 0  frame 0
        TX packets 143  bytes 39238 (38.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

By the way, GREAT PROJECT. I tried several manual ways to get an access point working and they all failed miserably.

billz, 2019-04-11:
This is a known issue with the RPi 3B's built-in Broadcom wifi module (brcmfmac). 

Specifically, the error `brcmfmac: brcmf_sdio_hostmail: Unknown mailbox data content: 0x40012` is a [bug in the Broadcom driver](https://github.com/lede-project/source/commit/c29a2a4283d71ae0c7b7d8fcbc27dbb1f94b3521) that uses a mailbox to communicate to the host. Refer to this issue for some possible workarounds https://github.com/raspberrypi/linux/issues/1342

This is not caused by RaspAP.

johan-gorter, 2019-04-11:
Thanks for the info! I looked on the internet to find the issue myself, but was unable to find this one.

The fix that I used was to change: `wpa_pairwise=TKIP ` to `rsn_pairwise=CCMP` in `/etc/hostapd.conf `. It seems to be running fine now.

I know that RaspAP is not causing this crash, but it may make RaspAP easier to use if RaspAP detects the given hardware and driver version and changes hostapd like this?

closed 2019-04-11 by johan-gorter

johan-gorter, 2019-06-11:
After trying more solutions, I could not get the access point to become stable on the raspberry pi 3B. It runs a couple of days at most before dying. I switched to a raspberry pi 3B+ and everything works smoothly now.

cedced19, 2019-08-19:
Is there any news with this issue ?