RaspAP/raspap-webgui #316: Hostapd not starting after plex installation.

alialiwa2005, 2019-04-14:
### Subject of the issue
Hostapd failing to initiate/activate.
### Your environment
* Raspberry Pi 3 model B
* Raspbian Stretch (9)
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* RaspAP Quick Install or Manual setup? Quick Install
* Using default configuration? (Y)
* Simultaneous AP and managed mode? (Y)
* Other software or services running with RaspAP? Plex media server/openvpn

### Steps to reproduce
Flash Raspbian Stretch Desktop Image to 32 GB microSD card via Etcher
Insert microSD into RPI 3
Connect via Ethernet to modem/router
Enable SSH
Update/Dist-Upgrade
Install RaspAP via quick install
Download and Install Plex Media Server (plex.tv) for Linux, "Ubuntu/Debian 64-Bit ARM v7"
Enable hotspot start at boot via crontab: https://github.com/billz/raspap-webgui/issues/200 (Issue 200)
Add lines to /etc/rc.local just before `exit 0`
```
service hostapd stop
sleep 5
service hostapd start
```
 (https://github.com/billz/raspap-webgui/issues/202#issuecomment-419394649)
Reboot.

### Expected behavior
After reboot, Hostapd should start automatically, and broadcast the wifi network from the AP.

### Actual behavior
After reboot, Hotspot wasn't running. Even when I manually enable Hostapd from webgui, it fails to start.

alialiwa2005, 2019-04-14:
I'd appreciate any help.
Also, as some routers have the option of integrating Plex, I'm sure it is possible to and that many would be happy if Plex media server is added natively in an upcoming version in the future.

billz, 2019-04-14:
> Other software or services running with RaspAP? Plex media server/openvpn

Per the [wiki](https://github.com/billz/raspap-webgui/wiki/Reporting-issues#supported-devices) these packages are not supported.

closed 2019-04-14 by billz

wsmeyer90277, 2019-04-27:
> 
> 
> > Other software or services running with RaspAP? Plex media server/openvpn
> 
> Per the [wiki](https://github.com/billz/raspap-webgui/wiki/Reporting-issues#supported-devices) these packages are not supported.

Are you saying RaspAP will not work with Plex Media Server?

I'm having a different issue in that Plex is unable to access the internet with RaspAP installed and this was the first Google result.

CrazyKamel, 2020-01-26:
> Are you saying RaspAP will not work with Plex Media Server?
>
> I'm having a different issue in that Plex is unable to access the internet with RaspAP installed and this was the first Google result.

Hi, I'm having the same issue with plex, I can't see any media. Do you find a solution ? Thank you.

billz, 2020-01-27:
The FAQ is your friend :)
https://github.com/billz/raspap-webgui/wiki/FAQs#iptables

wsmeyer90277, 2020-01-27:
Seriously?
The FAQ and your responses are ambiguous, that’s why we ask questions.

billz, 2020-01-27:
@wsmeyer90277 this FAQ directly addresses the Plex Media server.