RaspAP/raspap-webgui #317: Zero W No broadcast, No Errors

LibertyBeta, 2019-04-15:
### Subject of the issue
After Quick install, Zero W doesn't broadcast.

### Your environment
* Raspberry Pi hardware: Zero W
* Raspbian version Stretch Lite
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install or Manual setup? Quick
* Using default configuration? Y
* Simultaneous AP and managed mode? N/A
* Other software or services running with RaspAP? USB OTG.

### Steps to reproduce
Update. 
Install
Reboot. 

### Expected behavior
AP Setup Boardcast

### Actual behavior
Nothing happened, and we have no error logs that I can find.

billz, 2019-04-15:
The Pi Zero W needs to be configured as a managed mode AP via the UI _before_ a reboot.

LibertyBeta, 2019-04-15:
Instructions on mounting the UI before a reboot?

Or means to recover from that?

Also, as far as I can tell it never started broadcasting.

billz, 2019-04-15:
The UI is accessible in the browser via the `wlan0` IP address. The AP does not need to be active. Use `hostname -I` to find your IP.

LibertyBeta, 2019-04-15:
So, after doing this, and checking my configs, I've got errors(thanks be to OTG mode). 

No matter what I do it appears the uap0 interface isn't coming up. Logs to follow: 
```
-- Unit hostapd.service has begun starting up.
Apr 15 17:12:51 raspberrypi hostapd[3550]: Configuration file: /etc/hostapd/hostapd.conf
Apr 15 17:12:51 raspberrypi hostapd[3550]: Could not read interface uap0 flags: No such device
Apr 15 17:12:51 raspberrypi hostapd[3550]: nl80211: Driver does not support authentication/association or connect commands
Apr 15 17:12:51 raspberrypi hostapd[3550]: nl80211: deinit ifname=uap0 disabled_11b_rates=0
Apr 15 17:12:51 raspberrypi hostapd[3550]: Could not read interface uap0 flags: No such device
Apr 15 17:12:51 raspberrypi hostapd[3550]: nl80211 driver initialization failed.
Apr 15 17:12:51 raspberrypi hostapd[3550]: uap0: interface state UNINITIALIZED->DISABLED
Apr 15 17:12:51 raspberrypi hostapd[3550]: uap0: AP-DISABLED
Apr 15 17:12:51 raspberrypi hostapd[3550]: hostapd_free_hapd_data: Interface uap0 wasn't started
Apr 15 17:12:51 raspberrypi systemd[1]: hostapd.service: Control process exited, code=exited status=1
Apr 15 17:12:51 raspberrypi systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
```

billz, 2019-04-15:
Ah, you're using USB OTG. A more common setup involves preconfiguring `wpa_supplicant` to connect to the Pi Zero headless over WiFi, as described in the FAQ.

It appears in your setup `wlan0` has no configuration, so `uap0` cannot be brought up. Rather than troubleshoot this, I suggest you either follow the instructions in the FAQ, or use the manual installation steps with this project's `uap0` managed mode config.

billz, 2019-04-16:
Alternatively, configure your Pi Zero as a Wifi client using the UI. With `wlan0` configured, you can then start the AP in managed mode. Hope this clarifies things.

billz, 2019-04-22:
Closing due to inactivity. Feel free to update if your issue persists.

closed 2019-04-22 by billz