RaspAP/raspap-webgui #318: Auto start hotspot on boot

enannos, 2019-04-16:
Hello folks,

I know it is similar to #234 , but I cannot make it work on a PI Zero. 
It is really important to have the AP activated on boot, but I am unable to make it work.

I have tried with the service hostapd start in the rc.local but with no success.

Any ideas?

billz, 2019-04-19:
The Pi Zero W needs some extra steps that aren't covered in #234. There are a few guides online for doing this, however much of the information is outdated. So there's a good case for supporting it. 

I've made some commits to the project that add a systemd `raspap.service`. This uses the existing `servicestart` script to detect a Wifi client AP, add the `uap0` interface and start up network services in a predictable way on boot.

It's not enabled by default (yet). You can test this by running the Quick Installer again, enable [Wifi client AP mode](https://github.com/billz/raspap-webgui#simultaneous-ap-and-wifi-client), then execute the following:

```
sudo systemctl enable raspap.service
```
The AP should be active on reboot. Feedback is welcome.

billz, 2019-04-29:
Added to the Quick Installer and updated the FAQ. 
https://github.com/billz/raspap-webgui/wiki/FAQs#raspap-control-panel-works-but-there-is-no-wifi-after-reboot

Marking this as resolved.

closed 2019-04-29 by billz