RaspAP/raspap-webgui #319: Hotspot created on same channel/band managed wlan0

clean-ev, 2019-04-18:
### Subject of the issue
Wlan0 band/channel is used for AP 

### Your environment
* Raspberry Pi 3B+
* Raspbian version Stretch Lite
* Followed the project prerequisites? (Y/N) Y
* Checked the project FAQ? (Y/N) Y
* RaspAP Quick Install or Manual setup? Quick install
* Using default configuration? (Y/N) Y
* Simultaneous AP and managed mode? (Y/N) Y
* Other software or services running with RaspAP? None

### Steps to reproduce
pi@ApWiFi:~ $ iw uap0 info
Interface uap0
        ifindex 5
        wdev 0x5
        addr b8:27:eb:b7:d4:7f
        ssid raspi-we Gugi
        type AP
        wiphy 0
        channel 40 (5200 MHz), width: 80 MHz, center1: 5210 MHz
        txpower 31.00 dBm
pi@ApWiFi:~ $ iw wlan0 info
Interface wlan0
        ifindex 3
        wdev 0x1
        addr b8:27:eb:b7:d4:7f
        ssid Vighnaharta
        type managed
        wiphy 0
        channel 40 (5200 MHz), width: 80 MHz, center1: 5210 MHz
        txpower 31.00 dBm
pi@ApWiFi:~ $ cat /etc/hostapd/hostapd.conf
driver=nl80211
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
auth_algs=1
wpa_key_mgmt=WPA-PSK
beacon_int=100
ssid=raspi-webgui 
channel=11
hw_mode=g
ieee80211n=0
wpa_passphrase=ChangeMe
interface=uap0
wpa=2
wpa_pairwise=CCMP
country_code=US
ignore_broadcast_ssid=0

### Expected behavior
Expected AP to use band g and channel 11 per hostapd 
Unless this is not allowed by the hardware
### Actual behavior
AP band/channel follows what is used by wlan 0 as shown in output above

billz, 2019-04-19:
Not a bug. Check the output of `iw list`:

```
valid interface combinations:
    * #{ managed } <= 1, #{ P2P-device } <= 1, #{ P2P-client, P2P-GO } <= 1,
        total <= 3, #channels <= 2
    * #{ managed } <= 1, #{ AP } <= 1, #{ P2P-client } <= 1, #{ P2P-device } <= 1,
        total <= 4, #channels <= 1
```
The second valid combination indicates that a managed interface and AP is possible, but only on a single channel.

glaszig, 2019-08-02:
i think this can be closed?

billz, 2019-08-02:
yep, it sure can.

closed 2019-08-02 by billz