RaspAP/raspap-webgui #321: Add apple-touch-icon

mueller-ma, 2019-04-23:
### Subject of the issue
Currently the webgui doesn't include any apple-touch-icons. These icons are used if an OS (including iOS and Android) or Browsers need a high-res version of the favicon.

### Steps to reproduce
I pinned the website on the homescreen of an Android tablet and it looks pixelated.

### Expected behavior
The website should provide high-res icons.

### Actual behavior
It doesn't. You can use https://realfavicongenerator.net/ to generate these icons.

billz, 2019-04-23:
Great suggestion and long overdue. Thanks

mueller-ma, 2019-04-23:
Thank you for this project :)
I suggest you generate all possible icons to get the best result. For "Windows Metro" theme color use "Dark Red" as it's very close to the "RaspAP Red", but still one of the recommended colors.
Also add a version under "Version/Refresh". It used to be "2.0" a long time ago: https://github.com/billz/raspap-webgui/commit/3348bf7744b2c8ad81e89a7a44f2492b1a4bfc2c#diff-828e0013b8f3bc1bb22b4f57172b019dR77

commit a179ef05 (2019-04-23): Updated with hi-res icons. Resolves #321
closed 2019-04-23 by billz