RaspAP/raspap-webgui #322: Hotspot Running, no Internet Access

arrmo, 2019-04-26:
### Subject of the issue
Won't say this isn't me (likely is ... ;-)), but ... have things all connected, no traffic getting past the RPi => can access RaspAP configuration web pages from my client, but can't get traffic out. Running in managed mode, with AP enabled.

### Your environment
* Raspberry Pi hardware (examples: Pi 3 Model B, Pi Zero W)  
Pi 3B Plus

* Raspbian version (examples: Stretch Lite, Stretch Desktop)
Stretch Desktop

* Followed the project prerequisites? (Y/N)
Y

* Checked the project FAQ? (Y/N)
Y

* RaspAP Quick Install or Manual setup?
Install

* Using default configuration? (Y/N)
Y

* Simultaneous AP and managed mode? (Y/N)
Y

* Other software or services running with RaspAP?
Stock

### Steps to reproduce
As above, wlan0 is connected upstream (as client), all good. And AP is enable - also working, can connect to it, and browse RaspAP pages. Just no access past that.

### Expected behavior
Should be able to get past the RPi ... :-)

### Actual behavior
As above.

Thanks!!!

arrmo, 2019-04-26:
OK, I figured out the issue ... :-). The clients connected to the AP are getting the AP IP address as their DNS server (i.e. 192.168.50.1 if using the default). But the AP is not passing those DNS requests upstream ... are others seeing this?

If I do a name lookup on the RPi itself (e.g. google.com), NP! But from the client ... it fails. If I force the NS on the client (e.g. nslookup google.com 8.8.8.8) => that works. But without forcing the NS ... no joy.

Thoughts?

Thanks!

arrmo, 2019-04-27:
Resolved - removed and reinstalled dnsmasq, then working. So closing this out!

closed 2019-04-27 by arrmo

billz, 2019-04-27:
Good to know! Thanks for the debug info

mtdb, 2022-10-31:
I solved this problem by adding `interface wlan1 inet dhcp` at the end of the `/etc/dhcpcd.conf` file