RaspAP/raspap-webgui #324: DHCP for eth0

arrmo, 2019-04-29:
### Subject of the issue
My apologies in advance - this is more of a question than a bug ... at least for now ... LOL. I have checked the Wiki / FAQ, and can't seem to find this (may be me!), but - which side of the WAN / LAN router functionality is eth0 on? I'd like it to be on the LAN side - meaning => WAN = WiFi (wlan0), providing router functionality to WiFi (uap0) and wired ethernet (eth0). Is this possible? And I guess, in the end uap0 and eth0 are bridged, both providing DHCP to LAN clients, right?

### Your environment
* Raspberry Pi hardware (examples: Pi 3 Model B, Pi Zero W)  
Pi 3 Model B+

* Raspbian version (examples: Stretch Lite, Stretch Desktop)
Stretch Desktop

* Followed the project prerequisites? (Y/N)
Y

* Checked the project FAQ? (Y/N)
Y

* RaspAP Quick Install or Manual setup?
Quick Install

* Using default configuration? (Y/N)
N

* Simultaneous AP and managed mode? (Y/N)
Y

* Other software or services running with RaspAP?

### Steps to reproduce
As above, not sure it's an issue, until I can test it. Trying to understand it first.

### Expected behavior
As above.

### Actual behavior
TBD

voney, 2019-05-02:
I'm also trying to figure out how to set this up.

proffalken, 2019-06-11:
I'm also interested in this.

My use case is as follows:

I work away from home a lot, and use Raspap-Webgui as the bridge between my linux laptop and the hotel WiFi because (for some reason) this works better than connecting my laptop to the wifi directly.

I'm in the process of building out a "mobile IoT Lab" to take with me, and as part of this I'd like to have one pi running Raspap, in "dual-mode" (AP and Client), with DNSMasq running on eth0 *and* lo/uap0.

This would enable me to attach a switch to the ethernet port, and then share the wifi between the nodes in the "mobile lab" as well as wifi clients like my laptop.

I'm assuming that changes would need to be made that gave eth0 a static IP, and then reconfigured DNSMasq to listen on that interface as well, but I'm not 100% sure!

Thanks for an excellent product by the way, it's really seamless when I arrive at a hotel these days as a result!

manolosolalinde, 2019-07-12:
im also interested

HebusJon, 2019-07-15:
me to :)

olewsaa, 2020-01-28:
I have a fix for this, review my git page and wifi gateway page. 
https://github.com/olewsaa/Yacht-computer/tree/master/wifi2wifi/web-version
https://sites.google.com/site/olewsaa/yacht-server/raspberry-pi-as-a-router-gateway

It's working nicely. It is however sensitive to any web-gui clicks on save buttons.

glaszig, 2020-02-29:
there's currently no ui for this but recently #501 was merged. you can put custom configuration without fears of it being overwritten by raspap into `/etc/dnsmasq.d`.

in your cases the following should work.

```ini
# /etc/dnsmasq.d/eth0.conf

interface=eth0
dhcp-range=eth0,172.24.1.50,172.24.1.150,12h
```

adjust your ip ranges according to your `eth0` address space. something like this may also work with the `dhcp-host` option. consult `man dnsmasq`.

more info:

- https://jackywu.ca/enable-multiple-dhcp-range-for-multiple-nic-with-dnsmasq
- https://stackoverflow.com/q/9326438

mathojojo, 2020-04-30:
Good to see that this issue is Open again.
I successed to have something working, but with my poor knowledge, I'm almost sure it is not well implemented and there will probably bugs in IP traffics :/.

What I did : 
I modified 2 files.

- /etc/dhcpcd.conf

I added an eth0 par, that was totally absent.
```
# RaspAP eth0 configuration
interface eth0
static ip_address=192.168.0.1/24
static domain_name_server=9.9.9.9 1.1.1.1
```

- /etc/dnsmasq.d/090_raspap.conf
Under the existing wlan0 DHCP configuration, I set the eth0 DHCP server range there.
```
interface=eth0
dhcp-range=192.168.0.50,192.168.0.255,255.255.255.0,12
```
It's actually working for 20 hours. But I have some slow traffic sometimes (I'm not sure if it's a ISP network slow problem, or if it is due to my modifications.)

And it would be nice to bridge eth0 and wlan0 (br-lan ?) and make only one DHCP server for br-lan.

If someone can tell me if what I did is good, or better, if he know how I can do the bridge as describe above.

And of course even better, @billz, you could make it easy to set-up on the Raspap Web-GUI ?? It would be so GREAT.

Thank you in advance for the great help.

annunity, 2020-07-28:
Hi,

I tried doing what mathojojo did. But when I connect a device on eth1, I am not able to connect any devices on wlan0 since they cant get an ip address. Do I need to have different ranges for eth1 and wlan0 or should I bridge them? If so whats the best/easiest way?

Thanks for all the work.

billz, 2020-11-06:
This has been a popular request so I've made time to work on it. Follow the linked PR for progress updates.

[image]

billz, 2020-11-09:
Testing with an OrangePi3 (client) connected to RaspAP on `eth0` with an external adapter on `wlan1`:

[image]

billz, 2020-11-12:
Some notes:

1. A static IP address must first be configured for eth0 in **Networking > eth0**.
2. The **DHCP Server** settings page will detect a missing eth0 configuration and provide a message to this effect.
3. Enable the **DHCP for eth0** toggle, configure the eth0 interface and **Save settings**.
4. Be sure to define a different subnet for eth0 than the AP interface (192.168.0.1/24 is used here).
5. Reboot your device.
6. Connect a second device on eth0 (an orangepi3 is used here).
7. Monitor DHCP logs.

```
dnsmasq-dhcp[909]: DHCP, IP range 192.168.0.50 -- 192.168.0.255, lease time 12h
dnsmasq-dhcp[909]: DHCP packet received on eth0 which has no address
dnsmasq-dhcp[909]: DHCPDISCOVER(eth0) 192.168.0.253 02:▒▒▒▒▒▒▒▒▒▒▒
dnsmasq-dhcp[909]: DHCPOFFER(eth0) 192.168.0.253 02:▒▒▒▒▒▒▒▒▒▒▒
dnsmasq-dhcp[909]: DHCPREQUEST(eth0) 192.168.0.253 02:▒▒▒▒▒▒▒▒▒▒▒
dnsmasq-dhcp[909]: DHCPACK(eth0) 192.168.0.253 02:▒▒▒▒▒▒▒▒▒▒▒ orangepi3
```

At the moment the toggle lives on the **Advanced** tab while the interface is configured on the **Server settings** tab. Probably better UX to move the toggle to the latter.

darnokb, 2020-11-12:
@billz, any chance to make a choice between eth0 and eth1 or other eths in webgui?

billz, 2020-11-12:
@darnokb yes definitely. already thinking along those lines ;) I started with eth0 as a POC.

billz, 2020-11-16:
What began as a simple enhancement is turning into a complete refactoring of dhcp/dnsmasq 😅
Good things, though.

1.  DHCP option toggle for all interfaces
2. Server settings page is now ajaxified (big UX improvement)
3. Better & more consistent use of `/etc/dnsmasq.d`

[image]

RobertDuncan42, 2020-11-17:
Yeah. I have been trying to figure this out recently.  I think controlling from the DHCP Server page is the right way to go. 

Would it be possible to listen to eth(n) and if another DHCP service is detected disable the eth(n) DHCP on RaspAp?  This way plugging into my local LAN to service the Pi would just work.  Especially when I miss configure wlan0 and get locked out of RaspAp like I did last week.

billz, 2020-11-17:
@RobertDuncan42 I like the idea of detecting services on eth(n) and auto configuring it.

billz, 2020-11-24:
As the project has evolved more control over DHCP has moved from the **Networking** tabs to the **DHCP Server** UI. Interface .ini files are managed separately in `/etc/raspap/networking/` and written to `/etc/dhcpcd.conf`. A static IP must be defined before enabling DHCP on a given interface, and this is performed on a completely different UI in Networking, which then must be saved and applied. Many of the inputs in these two UIs are also redundant and can be confusing to the user. This UX could obviously be improved.

For sanity, I propose migrating control over the few remaining settings (fallback to static, static IP, interface metric, etc) to the DHCP server UI and deprecating the Networking interface tabs.

Nice thing about this approach is it a) centralizes everything DHCP-related in one area, and b) creates a space for future network-related enhancements (@zbchristian's networking > [mobile data UI](https://github.com/billz/raspap-webgui/issues/681), IP routing table, QoS, etc).

RobertDuncan42, 2020-11-25:
All network configuration on one panel, possible multiple tabs, makes sense. But it should be called Network Configuration, Not DHCP Server. So deprecate Networking and rename DHCP to Networking.

OzzieGaz, 2020-11-25:
I'm new here, but just love the work being done with RaspAP.  I have read a lot of documentation, some I understood, a lot a didn't, but wanted to ask a question that relates somewhat to the above discussion.

I would like to plug in an additional WiFi Access Point (probably a Unifi AP Pro) to the eth port on the Pi4, to extend or create a new WiFI SSID (or simply turn off the AP on the Pi4 and only use the Unifi AP).

If I simply plugged in the Unifi AP to the Pi4 eth port, is there any configuration I would have to do within RaspAP?  For example, enable dhcp on eth so that clients connecting via Unifi AP would be provided with an IP?

Thanks and apologies in advance if this has been asked before or is so simple, I can't see the wood for the trees!

billz, 2020-11-25:
@RobertDuncan42 I considered this, although the current DHCP server panel has management of static DHCP leases, active leases, upstream DHCP servers, etc. Could also be a place for MAC address filtering, which is often requested. Thinking of how to best group these settings in a way that makes sense to the end user.

ajkblue, 2020-12-05:
Would this work over a VPN connection too? I would like to use this so that any device connected over ethernet can have its traffic routed over a VPN as specified in the VPN Config page, just as can be down now over wifi.

billz, 2020-12-09:
@ajkblue the default behavior for VPN is to route traffic between tun0 and wlan0, but an option could be added to make the interface selectable.

ajkblue, 2020-12-10:
@billz It would be great if there would be an option so the eth0 interface could be used to route VPN traffic as well from either wlan0 or wlan1. This would be useful to use a VPN connection on devices where accessing the router isn't an option to plug in directly and the devices themselves don't have wifi cards. I hope that wouldn't be too difficult to implement.

billz, 2020-12-10:
@ajkblue sounds like a good 'ideas' topic for Discussions. let's keep the focus here on the issue.

billz, 2020-12-26:
The associated PR has been wrapped up with several enhancements + other fixes into pre-release v2.6-beta
https://github.com/billz/raspap-webgui/releases/tag/2.6-beta

Beta testers are welcome. See the release notes for info.

billz, 2021-01-25:
This feature is now part of the version 2.6 general release.

billz, 2024-02-08:
Thanks to everyone for their input in this issue.
For those on this thread that are interested in the related [WLAN Routing](https://docs.raspap.com/wlanrouting/) feature, add a quick reply and I'll send you an invite to the Insiders repo.

lnguyen234, 2024-04-05:
> Thanks to everyone for their input in this issue. For those on this thread that are interested in the related [WLAN Routing](https://docs.raspap.com/wlanrouting/) feature, add a quick reply and I'll send you an invite to the Insiders repo.

Hi, can I try the WLAN routing feature? thank you

bigbeka, 2024-05-31:
@billz thank you for a great piece of software! 
A questions about [WLAN routing](https://docs.raspap.com/wlanrouting/):

Does it work `usb0 > eth0`:
- `usb0` is 4G hat, provides WAN
- `eth0` is onboard LAN port, has DHCP and assigns IP to LAN client(s)?

PR #704 (2020-12-26): Feature: dhcp for eth0
Work in progress. Tested successfully with an OrangePi3 (client) connected via `eth0` to an RPi4 (server) runnning RaspAP. Relevant log entries below. Closes #324 

```
CLIENT -> DHCPDISCOVER
SERVER -> DHCPOFFER
CLIENT -> DHCPREQUEST
SERVER -> DHCPACK
```

```
dnsmasq-dhcp[1105]: 2721026064 available DHCP range: 192.168.0.50 -- 192.168.0.255
dnsmasq-dhcp[1105]: 2721026064 client provides name: orangepi3
dnsmasq-dhcp[1105]: 2721026064 DHCPDISCOVER(eth0) 172.16.10.33 02:07:4e:17:40:29 
dnsmasq-dhcp[1105]: 2721026064 tags: eth0
dnsmasq-dhcp[1105]: 2721026064 DHCPOFFER(eth0) [... cut]