RaspAP/raspap-webgui #325: wlan0 not getting IPv4 Address

arrmo, 2019-05-02:
### Subject of the issue
I have noticed wlan0 dropping some, and reconnecting - that's fine ... but it doesn't seem to be getting and IPv4 address, without manually running dhclient?

### Your environment
* Raspberry Pi hardware (examples: Pi 3 Model B, Pi Zero W)  
Pi 3 Model B+

* Raspbian version (examples: Stretch Lite, Stretch Desktop)
Stretch Desktop

* Followed the project prerequisites? (Y/N)
Y

* Checked the project FAQ? (Y/N)
Y

* RaspAP Quick Install or Manual setup?
Quick Install

* Using default configuration? (Y/N)
Y

* Simultaneous AP and managed mode? (Y/N)
Y

* Other software or services running with RaspAP?

### Steps to reproduce
Reboot, check wlan => connected, but only an the RaspAP interface shows "No IPv4 Address Found" ... which is right, see below,

```
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet6 fe80::ba27:ebff:fe54:870d  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:54:87:0d  txqueuelen 1000  (Ethernet)
        RX packets 9843  bytes 1327880 (1.2 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2095  bytes 613688 (599.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

### Expected behavior
Expecting an IPv4 address assigned.

### Actual behavior
As above, only inet6 address. But .. if I run,
sudo dhclient wlan0, then all is good,
```
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.2.200  netmask 255.255.255.0  broadcast 192.168.2.255
        inet6 fe80::ba27:ebff:fe54:870d  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:54:87:0d  txqueuelen 1000  (Ethernet)
        RX packets 11095  bytes 1540548 (1.4 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 3103  bytes 957829 (935.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

billz, 2019-05-02:
That's odd. Tested on a freshly installed Pi Zero W. The only other difference is Stretch Lite rather than Desktop. wlan0 after reboot shows a good IPv4 address:

```
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.40  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::ba27:ebff:fe13:e81b  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:13:e8:1b  txqueuelen 1000  (Ethernet)
        RX packets 69  bytes 15914 (15.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 91  bytes 15379 (15.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

Suggest you try to repro this with Stretch Lite. Is there a reason you're running Desktop?

arrmo, 2019-05-03:
Will try resetting everything first (need to search the docs, see if there is a clean way to reset all of raspap to defaults). I need Desktop because of VNC => setting this up for my daughter, at university ... so the only way in is a cloud type connection (i.e. behind firewall that I can't control). But I'm open to suggestions (to not use Desktop)!

Thanks!

arrmo, 2019-05-03:
Nope, pulling my hair out - arrgh! After a reboot, I have to manually run dhclient ... but I didn't before. Even after uninstalling the problem seems to remain, like a file has gotten messed up somehow.

Any suggestions (other than wiping the entire OS of course .. :-))?

billz, 2019-05-03:
Adding `sudo dhclient wlan0` to `/etc/rc.local/` should resolve this. Refer to the official RPi docs on how to edit this file. 

If not, run the [Quick Installer again ]( https://github.com/billz/raspap-webgui/wiki/Reporting-issues#will-raspap-let-me-create-a-configuration-that-breaks-my-hotspot) to restore the default project settings. 

All relevant config files are backed up to `/etc/raspap/` prior to running the Quick Installer, so you can check here as well.

arrmo, 2019-05-11:
Sorry, was out of town. Just got back in - that works, thanks! But I have to ask ... why is this needed (or is it added to your code now?)?

Thanks again.

billz, 2019-05-13:
> why is this needed (or is it added to your code now?)?

Glad the fix worked for you. This appears to be unique to Stretch Desktop, while this project recommends the Lite distribution.

closed 2019-05-13 by billz