RaspAP/raspap-webgui #327: ERR_CONNECTION_TIMEDOUT when trying to access config page on chrome

Razzorback225, 2019-05-12:
### Before submitting an issue

Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues. 

### Subject of the issue
After the installation was (or I tought it was) successfull and after a reboot, I tried to managed the hotspot with the manager web page but i can't connect on it due to an connection timed out error
### Your environment
*Raspberry Pi 3 B 
* Raspbian Stretch Lite 2019-04-08 v4.14
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* RaspAP Quick Install or Manual setup? Quick Install
* Using default configuration? (Y)
* Simultaneous AP and managed mode? (Y/N) Don't know what it is
* Other software or services running with RaspAP? Xserver, mono, mysql, kupiki

### Steps to reproduce
Just try to connect to the network manager page

### Expected behavior
Connection popu showing up

### Actual behavior
ERR_CONNECTION_TIMED_OUT page

billz, 2019-05-13:
Have you tried installing RaspAP on a clean setup? 

Kupiki installs its own hotspot, web frontend, captive portal and so on. It's likely this has created a conflict with RaspAP. Suggest you choose one or the other, not both.

closed 2019-05-13 by billz

Razzorback225, 2019-05-14:
Yep, working perfectly on clean install

Mangoz1x, 2024-12-25:
I just discovered this after so many frustrating hours, for some reason it wont load on chrome, but on safari, it works perfectly fine. To anyone who may discover this 5 years later. Just try a different browser.