RaspAP/raspap-webgui #329: Possibility of changing IP of the Hotspot

FabriceC01, 2019-05-15:
### Subject of the issue

I Would like to change the IP address of my Hotspot (ex : having 192.168.1.1 instead of 10.3.141.1)
I know how to do by modifying dhcpd.conf and dnsmasq.conf manualy but if I do a modification 
such as SSID with de webgui after, the initial IP 10.3.141.1 is coming back.

Is it possible to change it somewhere permanently ?

Thanks

Fabrice
 
### Your environment
* Raspberry Pi hardware Pi 3 Model B+ 
* Raspbian version Stretch Desktop
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* RaspAP Manual setup
* Using default configuration? (N)
* Simultaneous AP and managed mode? (N)

FabriceC01, 2019-05-20:
Hello,
Nobody has an idea to help me ?
Thanks.
Fabrice

billz, 2019-05-27:
@FabriceC01 sounds like a bug, I'll look into it. Thanks for reporting.

FabriceC01, 2019-05-28:
Hello,
Thanks for your help. Do you need my conf files ? they're enclosed.

The greatest would be to be able of changing the AP IP in the webui.
More than that : I defined a bridge between WiFi and ethernet (see joined file).  it's not possible to see the dhcp ranges correctly with you're tool...

thanks for all.

Fabrice

[dhcpcd.conf.txt](https://github.com/billz/raspap-webgui/files/3225799/dhcpcd.conf.txt)
[dnsmasq.conf.txt](https://github.com/billz/raspap-webgui/files/3225801/dnsmasq.conf.txt)
[hostapd.conf.txt](https://github.com/billz/raspap-webgui/files/3225802/hostapd.conf.txt)
[interfaces.conf.txt](https://github.com/billz/raspap-webgui/files/3225803/interfaces.conf.txt)
[sysctl.conf.txt](https://github.com/billz/raspap-webgui/files/3225804/sysctl.conf.txt)

garyng, 2019-06-20:
Can reproduce. 
Any updates for this issue?

Thanks!

glaszig, 2019-08-02:
this is currently a limitation. as soon as you touch and save your hotspot config, either of those two static ip address (`10.3.141.1/24`, `192.168.50.1/24`) [will be written to the configuration](https://github.com/billz/raspap-webgui/blob/2ded7be76ba878b8284ce36abf6d4c589d0e19da/includes/hostapd.php#L649-L691).

FabriceC01, 2019-08-20:
Hello,
Thanks glaszig for your comment.
I have neither the time nor the skills to make the enhancement but I know now where to change the IP in the php. 
Fabrice

geostant, 2019-09-02:
I can confirm that I've the same issue here
can't change from the WEBUI the DHCP range

billz, 2019-09-03:
> i prefer 2 as it leads to idempotent processes and is easier to manage.

Agreed. This 'fallback to default' was introduced with support for managed mode APs, which was a big enhancement. It needs to be updated to correctly parse settings from the Networking UI.

billz, 2019-09-08:
This commit correctly reads/sets values from the networking UI when saving settings in **Configure hotspot**. The exception, for the moment, is the special interface `uap0` which is created for the managed mode AP.

iandk, 2019-09-09:
Awesome, so setting a custom range should work now? 
Like 10.0.0.1 for the device itself and 10.0.0.50-10.0.0.100 as the DHCP range? 

Additionally it would be awesome if this would be possible for uap0 as well. 

Thanks!

iandk, 2019-09-23:
Any feedback on this?

Thanks, Ian

billz, 2019-11-04:
These commits apply the user-defined static IP address to the `uap0` interface. In addition, the `dhcp-option` value '6' is correctly applied, so this handles custom DNS servers for client name resolution as well.

closed 2019-11-04 by billz