RaspAP/raspap-webgui #330: 2x wifi adapters

symbios24, 2019-05-16:
Hello there

i wanted to ask is your program supporting to have 2x usb adapters? 
i mean to use one wifi adapter to connect to another hotspot to take internet and use the other one to create a hotspot to share this internet?

Thanks

billz, 2019-05-17:
>  is your program supporting to have 2x usb adapters?

Indirectly, yes. That is, you may create any number of interfaces from the command line and administer them with the **Configure networking** UI. 

However, why not just use the onboard Wifi? You can use RaspAP to connect to an existing hotspot (your router, for example) and share an internet connection with a managed mode AP.

symbios24, 2019-05-18:
Thanks for the response i have a raspberry pi 2 thats why
can you put an sd card image for download ?

Thanks

billz, 2019-05-18:
The UI doesn't create interfaces, apart from `uap0` which is used for a managed mode AP. You can however manage existing interfaces from the UI. 

I'd suggest you look at this [FAQ](https://github.com/billz/raspap-webgui/wiki/FAQs#can-i-use-wlan0-and-wlan1-rather-than-eth0-for-my-ap) which describes a process for using wlan0 and wlan1. 

Closing this as there is no bug report here. Thanks.

closed 2019-05-18 by billz