RaspAP/raspap-webgui #334: PiZeroW: Can't scan for networks while in AP mode

newhinton, 2019-06-11:
Pi Zero W
Stretch Lite
Using default configuration from the quickinstaller

I want to use this as a "setup" for this device. Which means that per default, raspap opens a wifi-hotspot, so that someone can connect and set it up to use his homenetwork.  ( I will write a script that starts the hotspot again if the configured clientconfig does not work for whatever reason)

However: While the AP is open, i cannot scan for results in client mode. 
Also, i cannot connect to it, it fails with "cannot execute 'wpa_cli reconfigure'"
I have done my research, and it seems that wpa_cli is not able to access wlan0 while the hotspot is running. I have tried [this](https://unix.stackexchange.com/questions/415816/i-am-trying-to-connect-to-wifi-using-wpa-cli-set-network-command-but-it-always-r), but when i do so, it kills the ap. 

Is this a known bug, a hardware limitation or simply working as intended?

Maybe it would be an idea to allow for manually entering the ssid and the password, like android does, so that a scan would not be nessessary

newhinton, 2019-06-26:
Edit: This also affects the new Raspberry Pi 4b with Buster Lite

glaszig, 2019-08-02:
could you please post the output of `iw list` for both your pi's here so that we can see hardware capabilties? thank you.

billz, 2019-08-17:
@newhinton I configured a Pi Zero W as an AP in managed mode, connected several clients, and have no issues scanning for networks.

As @glaszig mentioned, the output of `iw list` is useful here. For example, the output on your Pi should include the following:

```
valid interface combinations:
    * #{ managed } <= 1, #{ P2P-device } <= 1, #{ P2P-client, P2P-GO } <= 1, total <= 3, #channels <= 2
    * #{ managed } <= 1, #{ AP } <= 1, #{ P2P-client } <= 1, #{ P2P-device } <= 1, total <= 4, #channels <= 1
```

Are you using the onboard wireless of the Pi Zero, or an external adapter?

newhinton, 2019-08-17:
hm strange

i am using onboard, i will try again tomorrow, fully from scratch. maybe ill find hints then why it does not work for me

glaszig, 2019-08-17:
> i am using onboard

huh?

newhinton, 2019-08-17:
ah sorry, i am using the inbuilt wifichip, not an external dongle

newhinton, 2019-08-28:
Hm, i have now used a later version of raspbian, and reinstalled everything. Now it works (at least for the pi4, haven't tried the zero yet)

Anyway here the result: 

<details><summary>Result:</summary>
<p>
```bash
 
Wiphy phy0
        max # scan SSIDs: 10
        max scan IEs length: 2048 bytes
        max # sched scan SSIDs: 16
        max # match sets: 16
        max # scan plans: 1
        max scan plan interval: 508
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports roaming.
        Device supports T-DLS.
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)                                                                                                      
                * TKIP (00-0f-ac:2)                                                                                                        
                * CCMP-128 (00-0f-ac:4)                                                                                                    
                * CMAC (00-0f-ac:6)                                                                                                        
        Available Antennas: TX 0 RX 0                                                                                                      
        Supported interface modes:                                                                                                         
                 * IBSS                                                                                                                    
                 * managed                                                                                                                 
                 * AP                                                                                                                      
                 * P2P-client                                                                                                              
                 * P2P-GO                                                                                                                  
                 * P2P-device                                                                                                                                                                                     
        Band 1:                                                                                                                                                                                                   
                Capabilities: 0x1022
                        HT20/HT40
                        Static SM Power Save
                        RX HT20 SGI
                        No RX STBC
                        Max AMSDU length: 3839 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 16 usec (0x07)
                HT TX/RX MCS rate indexes supported: 0-7
                Bitrates (non-HT):
                        * 1.0 Mbps
                        * 2.0 Mbps (short preamble supported)
                        * 5.5 Mbps (short preamble supported)
                        * 11.0 Mbps (short preamble supported)
                        * 6.0 Mbps
                        * 9.0 Mbps
                        * 12.0 Mbps
                        * 18.0 Mbps
                        * 24.0 Mbps
                        * 36.0 Mbps
                        * 48.0 Mbps
                        * 54.0 Mbps
                Frequencies:
                        * 2412 MHz [1] (20.0 dBm)
                        * 2417 MHz [2] (20.0 dBm)
                        * 2422 MHz [3] (20.0 dBm)
                        * 2427 MHz [4] (20.0 dBm)
                        * 2432 MHz [5] (20.0 dBm)
                        * 2437 MHz [6] (20.0 dBm)
                        * 2442 MHz [7] (20.0 dBm)
                        * 2447 MHz [8] (20.0 dBm)
                        * 2452 MHz [9] (20.0 dBm)
                        * 2457 MHz [10] (20.0 dBm)
[... 100 lines cut ...]
                 * update_connect_params
        Supported TX frame types:
                 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
        Supported RX frame types:
                 * managed: 0x40 0xd0
                 * P2P-client: 0x40 0xd0
                 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * P2P-device: 0x40 0xd0
        software interface modes (can always be added):
        valid interface combinations:
                 * #{ managed } <= 1, #{ P2P-device } <= 1, #{ P2P-client, P2P-GO } <= 1,
                   total <= 3, #channels <= 2
                 * #{ managed } <= 1, #{ AP } <= 1, #{ P2P-client } <= 1, #{ P2P-device } <= 1,
                   total <= 4, #channels <= 1
        Device supports scan flush.
        Device supports randomizing MAC-addr in sched scans.
        Supported extended features:
                * [ 4WAY_HANDSHAKE_STA_PSK ]: 4-way handshake with PSK in station mode
                * [ 4WAY_HANDSHAKE_STA_1X ]: 4-way handshake with 802.1X in station mode
```

</p>

billz, 2019-08-28:
The project [prerequisites](https://github.com/billz/raspap-webgui#prerequisites) do specify first updating Raspbian to the latest version.

newhinton, 2019-08-28:
Yes, and i was using the latest raspbian version at that time, that's why i was confused :D but gladly, it works now :)

closed 2019-08-28 by newhinton