RaspAP/raspap-webgui #335: wlan0 and wlan1 configuration

symbios24, 2019-06-18:
Hi, i did this: https://github.com/billz/raspap-webgui/wiki/FAQs#can-i-use-wlan0-and-wlan1-rather-than-eth0-for-my-ap

so i have two wifi available (internal and usb external) and use one for client and the other for hotspot
and i cannot access the web interface anymore to from the same raspberry to configure the wifi adapters,,before i could do it

domenicomazza, 2019-06-19:
Try connecting your machine to the RaspAP network and try it's router address 10.3.141.1 (you'll see it's set in dhcpd.conf).

I am also trying to get wlan0 and wlan1. Thank you for the link... I've been looking everywhere and tried so many different things only for it to be in the FAQ the whole time. I've followed the steps I have wlan0 which acts as the AP and wlan1 which successfully gets onto my main network now (the furthest I have reached to get a wire-free repeater).. yet I still cannot get a web connection. I'd love to know what to debug here and what logs I can look at? I'm guessing it must be the DNS. I have it all setup with 1.1.1.1 and 1.0.0.1 (CloudFlare DNS). I'm on an RPi 1 B+ with two USB wifi dongles (drivers installed).

carlosanunes, 2019-06-20:
Hi,

Same here. Trying to use a RPi Zero W as a pure wireless router (wlan0 to wlan1, wlan0 provides the internal wifi, wlan1 connects to the internet wifi connection).

Hardware:
RPi Zero W
Wifi Dongle

After experimentation (trial and error), my conclusion is that the FAQ does not actually refer to this, or at least the method does not work or is outdated.

Steps:

1. Stash any current wifi connection config.

sudo cp /etc/wpa_supplicant/wpa_supplicant.conf /etc/wpa_supplicant/wpa_supplicant.conf.sav
sudo cp /dev/null /etc/wpa_supplicant/wpa_supplicant.conf

2. Install RaspAp and Reboot

sudo wget -q https://git.io/voEUQ -O /tmp/raspap && bash /tmp/raspap

(after reboot, the new WiFi network created by RaspApp should be available, connect to it)

3. Re-enable WiFi Config

sudo cp /etc/wpa_supplicant/wpa_supplicant.conf.sav /etc/wpa_supplicant/wpa_supplicant.conf

4. Reboot Again

sudo reboot

(at this point wlan0 should be providing the internal network, and wlan1 should be connected to the network defined in the credentials in wpa_supplicant.conf)

5. Make wlan0 and wlan1 communicate with each other

And this is the step I am missing. If I "ping -I <name of wlan1 interface> 8.8.8.8" I get a connection, so every adapter is doing what is supposed to. The issue here is make wlan0 re-route traffic from the internal network to the external one, wlan1. Undoubtedly there are iptable commands to achieve this (and will eventually find them), but a more ui friendly approach would be nice. :)

domenicomazza, 2019-06-20:
I've given up and purchased a second hand unit for my exisiting mesh system. The RaspAP software is brilliantly setup to get eth0 to wlan0 working straight out of the box. Everywhere I read that suggested wlan0 to wlan1 is possible says 'do this and this to get two wireless adapters to work' and it just doesn't work. Then there are others that state this is not meant to be possible because of a specification e.g. https://raspberrypi.stackexchange.com/a/83450

On as side note: I have managed to get one thing working—a bridge to allow RaspAP to use the same IP as my main router ... I used this guide https://blog.ithasu.org/2016/10/using-a-raspberry-pi-3-as-a-wifi-access-point-and-bridge/ .. just beware RaspAP will reset hostapd.conf if you change any AP settings like your password or SSID from the web interface.

carlosanunes, 2019-06-20:
I think we are trying to achieve different goals... so maybe the issue first is better documentation. :)

I am going to open a new issue for my own use case as this one is more general.

glaszig, 2019-07-22:
the problem is that there‘s currently only one wpa_supplicamt configuration for any and all (additional) wifi adapters. yesterday i plugged in a ralink usb dongle and bang, it git connected to the same wifi network.

what the ui needs is a way to select which interface to connect with which wifi network. then create different configs for the interfaces.

de-mon, 2019-07-25:
I found this question and answer: 
https://superuser.com/questions/469039/wpa-supplicant-for-dual-wlan
Which contains: 
`create a new file in directory /etc/wpa_supplicant/. The new file should be named with interface name i.e

wpa_supplicant-wlan1.conf

file /etc/wpa_supplicant/wpa_supplicant-wlan0.conf for interface wlan0
file /etc/wpa_supplicant/wpa_supplicant-wlan1.conf for interface wlan1`
I haven't had an opportunity to try this yet, but I would hope changing /etc/wpa_supplicant.conf to wpa_supplicant-wlan1.conf would work? Will test and report back.

Darkw00, 2019-07-31:
I managed to get wlan0 and wlan1 to work by following this guide:
https://github.com/billz/raspap-webgui/issues/198

One thing that i needed to add was to comment out "static routers=" line in /etc/dhcpcd.conf

feh123, 2019-08-01:
I have a raspberry pi 3B running buster lite. My usb wifi adaptor is the recommended Edimax N150. I followed the quick install guide and enabled wlan1 and the AP client. The interface is up, hostapd is running - it all seems fine. However when I login to the hotspot I get the error no internet connection. I tried altering the hotspot - going from g to n, changing the security but no fix. I also tried #198. I do not have a line 'static routers=' in my /etc/dhcpcd.conf but I have made all the changes suggested in #198. I seem close but something is still missing. Any ideas much appreciated.

galorin, 2019-09-30:
@feh123 I have had similar when using wlan0 as the AP and wlan1 as the bridge interface.

It looks like RaspAP is creating two default routes and traffic is getting confused.

What I needed to do (and need to do on every reboot) is delete all default routes and then manually add the default route.  The quick script I use is below, and it is not that good.  Unfortunately, my Pi is not just an AP, so I am running this when I ssh in to do other stuff.  I haven't done anything about running this when interfaces come up or something smarter.  It also needs re-running if something renews a DHCP lease, static or otherwise.

```
#!/bin/bash
sudo route del default #wlan0 default gateway
sudo route del default #remaining wlan1 default gateway
sudo route add default gw 192.168.1.254 wlan1
```

enodr, 2019-10-13:
When using a second wifi interface as a bridge (wlan1) the problem is that the metric for the route for wlan1 is higher than the one for wlan0. Adding a higher metric for wlan0 solves the problem:
edit /etc/dhcpcd.conf and add under:
interface wlan0
metric 999

eth0 works fine because when plugged it has a lower metric.

Best solution would be not to add the default route 0.0.0.0 for wlan0 when used in AP mode.

seandoran, 2019-10-30:
I am managing to have the exact same issues as @feh123 only with the edimax N150, using a tp-link tl-wn722n works fine without issue. I tried using the n150 on a raspi3b+ in buster desktop and noticed that the wlan1 was disassociated from the wifi network I wanted to bridge, so I logged in through the web-gui, removed the password from the wifi client page and rebooted. When i got back to the desktop i logged back in through the web-gui page, this gave me internet back on my raspi but as soon as I try to connect to the box from my phone it (the raspi) freezes and I have to restart. Are we sure the N150 should be recommended for this project? For the moment I'll use my tp-link to run things which is unfortunate because I need a less bulky build. Any advice helps 😁👍

feh123, 2019-10-30:
I have continued to work on this without a completely successful result. The major influencer is the chipset in the adaptor. So I have abandoned the N150 and had partial success with the Linksys WUSB6300 adaptor with the RTL8812AU chipset. This does need to be compiled but https://raspberrypi.stackexchange.com/questions/64502/install-drivers-for-rtl8812au-wireless-usb-adapter, works.
With this and a +3B I can get a partial success. There seems to be some issue with the order in which you do the installation following the FAQs for wlan1. I have not resolved this - I can get an AP to work but not everything will connect to it! However more often when I do the installation now (when I updated to 1.6.2) I do not get a full AP so only one device at a time can access the internet.
I have not done enough work to understand what is causing this but I feel there is a solution somewhere in this - possibly an issue with 1.6.2 but more likely there is a correct order to doing the installation steps I am missing. As an aside the Atheros AR9271 seems a very good chipset for AP but I have not tried this so far with Raspap.

up2smoke, 2019-12-08:
> I found this question and answer:
> https://superuser.com/questions/469039/wpa-supplicant-for-dual-wlan
> Which contains:
> `create a new file in directory /etc/wpa_supplicant/. The new file should be [...]

Very very Long Time u need to Test and Report back. Or it was a lie?

up2smoke, 2019-12-08:
@feh123 i have the same RTL8812AU chipset and a found a way how its can Work, but my english is bad, so i will only write the commands.
Starting with a clean install of RaspianLite on a rPi Zero W
> sudo apt update
> sudo apt upgrade -y

Install driver for 8812au
> sudo wget http://www.fars-robotics.net/install-wifi -O /usr/bin/install-wifi
> sudo chmod +x /usr/bin/install-wifi
> sudo install-wifi

Set 8812au driver as force wlan1
> sudo nano /etc/udev/rules.d/72-static-name.rules
`ACTION=="add", SUBSYSTEM=="net", DRIVERS=="brcmfmac", NAME="wlan1"`
> Sudo reboot

Install raspap
> curl -sL https://install.raspap.com | bash
> sudo reboot

After that i was checking the raspi-webgui
Everthink was working fine on wlan0 (onboard wifi - ip 10.3.141.50)
wlan1 (8812au - ip 192.168.178.27) was still conneced to the Router (192.168.178.1)

Open an SSH shell:
> sudo nano /etc/udev/rules.d/72-static-name.rules
Replace the "8812au" driver with "brcmfmac" and 
> sudo reboot

Thats the way how i get a Bridge between wlan0 and wlan1 
RaspAp Dashboard will Show that wlan0 is not connected, but when u connect to raspi-webgui it will works fine. No error with dhcp or that no internet is aviable.

Please report Back If that Work For u

feh123, 2019-12-08:
Hi @up2smoke , thanks for the ideas. I do not have a Pi zero available so I had to try it on a +3 B.

All went okay to checking raspi-webgui after adding raspap and rebooting. Although raspi-webgui SSID was up I could not connect to it. The linksys showed on my network (via fing) as a router. But wlan0 was there on 10.3.141.1 and wlan1 also gave an IP address. I could open the raspap web site but no connection. 
I changed from 8812au to brcmfmac and this altered my network - now the Pi appeared at it's usual IP address and no linksys router shows up. But still no connection to raspi-webgui.
I configured the hotspot to my usual values and enabled the AP. This made it worse - I now lost the hotspot SSID completely.  Hostapd and dnsmasq were both down. 
I am going to repeat in again from a fresh install to make sure I did not have a typo error. Thanks again for working on this.

up2smoke, 2019-12-08:
@feh123 
Please check
> lsmod
if rPi3+ use the same wifi driver for onboard chipset und If its not replace this, instand of "brcmfmac" -maybe u have a diffrent driver.

First step i forgot to tell u: when i make a clean install, i put a "wpa_supplicant.conf" in my /boot partition, coz i only run my rPi Zero headless. 
Maybe this step is importend to

I just Change the SSID and the Security to wpa2, coz i'm afraid thats not running anymore, when i Change to much :"D

feh123, 2019-12-08:
Okay I will. I put wpa_supplicant.conf in my boot as well. I also add ssh as I am running headless too.
I did leave out changing wlan0 to wlan1 in config.php and altering dhcpcd.conf as Raspap FAQs request. 
Should I change these? 
Also I guess you do not want AP enabled?

up2smoke, 2019-12-08:
Yes, u re right. I dont changing anythink. Everytime when i tried this, it was not possible to connect to the raspi-webui. All the time i got Error "to get IP" or there was no connection to the world wide web. 
After u Install 8812au Driver u will See that both (onboard & external) to ur router. 
After Install RaspAp and open 10.3.141.1 u will see that RaspAp runs on onboard wifi and external wifi will be connected to ur router. But u still dont have Connection to the Internet. Thats the reason why i was testing whats happen when i change the the driver for wlan1 to force onboard chipset. After reboot i can See: wlan0 (extnernal WiFi) is the raspi-webui (ip10.3.141.1) and wlan1(onboard wifi) is connected to the router.
Dashboard shows green "wlan0 Up" but no connection. But it is. And Internet works but slow

feh123, 2019-12-08:
Hi @up2smoke - this time by only changing security and name for SSID it worked! Thanks.
Just one question - if I need to change the client wifi ssid and password to connect to a different wifi how do I do it?
Thanks again for the procedure.

up2smoke, 2019-12-08:
@feh123 nice to listen that. Maybe the dev of RaspAp have now an Idea how to handle a bridge between wlan0 & wlan1..

I dont know how to chnage the wifi.
My wpa_supplicant looks like this:

country=DE
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
ap_scan=1
network={
	    ssid="test001"
	psk="****"
	key_mgmt=WPA-PSK
	priority=10
}
network={
	ssid="test002"
	psk="****"
	key_mgmt=WPA-PSK
	priority=1
}
network={
	ssid="test003"
	psk="****"
	key_mgmt=WPA-PSK
	priority=2
}

Thats means: test001 is the hotspot from my Phone, test002 the router at my home and test003 the wifi from my girlfriends House. Imoptend is the priority !!!
A high number will matched First, thats why i have test001 the priority 10. I use the mobile Hotspot Like a Fall Back Option, coz my Phone Tell me in one second if one client is connected or Not, with the right ip.
Also its Work, when pi is connected to test002 or test003 and test001 cums online..he Change to test001 coz this have the highest priority

feh123, 2019-12-08:
Thanks @up2smoke . I will try the multi-wifi configuration, add it to boot on my sd card from my laptop and see if Raspap likes it! I will let you know.

feh123, 2019-12-09:
The multi-SSID wpa with priority worked fine. The setup works very well on a +3B.

So if Raspap could include this bridge that would be great. Thanks for all your help!

feh123, 2019-12-09:
I often use my RPI Raspap when travelling. One idea for an enhancement would be to allow a second RPI to use the wifi from the first RPI and re-broadcast either the same SSID or a new SSID. 
This would cover the situation when the distance is large and the RPI wifi range is limited. The basis of a mesh wifi? I will add this as a separate enhancement idea.

enterprisebug, 2020-01-28:
may i get some ideas:
- when i am sitting in front of my 3B+ i see it is connected to the internet (wlan1) *and* provides an AP (wlan0). Also i can browse websites on 3B+
- i am able to connect with the AP from another device. e.g. my mobile phone or laptop.

problem:
- from my understanding all i need to is is to "connect/passthrough" wlan0 and wlan1 somehow.

where do i have to do a magical configuration to get this working?

enterprisebug, 2020-01-29:
> may i get some ideas:
>
> * when i am sitting in front of my 3B+ i see it is connected to the internet (wlan1) _and_ provides an AP (wlan0). Also i can browse websites on 3B+
> * i am able to connect [...]

i was able to get it working by just doing:

    sudo systemctl enable raspap.service

PidgeyBE, 2020-02-21:
I got it working, with wlan1 as client and wlan2 as AP. This is what I did:
* Install RaspAP via quick installer.
* Start hotspot in the RaspAP UI on wlan2
* From this point onwards, *don't touch the UI anymore*
* edit `/etc/hostapd/hostapd.conf` as you wish. I'm using 5GHz for my AP and 2.4GhZ for client connection for example, to avoid interference! Look online to know how what the stuff in hostapd.conf means. Definitely also check out the `ht_capab` parameter. Via `iw list` you can find out what HW capabilities your wlan adapter has.
* Edit `/etc/dhcpcd.conf` and only put the following in it:
```
interface wlan2
static ip_address=10.3.141.1/24
static domain_name_servers=1.1.1.1
nohook wpa_supplicant
```
* Set up your client connection via wpa-supplicant.
* Reboot 
* Enjoy!

MomoSho, 2020-05-13:
Hello, I have an incomplete understanding about how networking and routing works so forgive me if my question is misplaced, as I believe it is apart of this particular discussion. I'm trying to clarify if I'm understanding things correctly.
I have a pi 3 B+, which comes with a WiFi adapter on board. I want to be able to use the Pi as an access point and I think a WiFi client. I assume a WiFi client connects to a router to access the internet. 
The part I want to clarify, which is probably a stupid question, but, in order to enable both the AP and WiFi client, are two network adapters required, or can the Pi's on board network adapter be both the AP (wlan0) and the WiFi Client (wlan1)? Will I need to either connect the Pi to the network via Ethernet or a USB WiFi dongle to access both functionalities simultaneously?
Thanks

PidgeyBE, 2020-05-13:
Indeed, the internal wifi chip can act both as client and AP at the same time. However, in my experience this doesn't work well, probably because of interference of making the chip do things it's not really build for...
So it's better to have an extra external USB WiFi dongle, preferably a 5GHz one, so one chip can work on 5GHz and the other on 2.4GHz... To eliminate the interference. That's what works best in my experience.

billz, 2020-05-13:
@MomoSho excellent question, actually. @PidgeyBE is correct; the internal wifi chipset is able to operate in AP-STA mode, ie. simultaneous wireless client and AP. While it is entirely functional, it is less than optimal compared with an external adapter, mainly due to radio interference.

The Wifi client AP mode button is enabled after you configure a wireless client.

MomoSho, 2020-05-19:
Great, thanks guys! I was able to get it working using just the built in WiFi chipset (or additional external WiFi dongle). My WiFi network is 5G and the AP is running on 2.4G (if I understand that correctly).
I was misunderstanding the FAQ for too long on _when_ to enable the AP (uap0) with the WiFi client (wlan0).
Before I always already had my WiFi already configured, but my misunderstanding was that with the WiFi configured, install RaspAP, then open the browser to the localhost (or local IP) to enable the AP with the WiFi client, _then_ reboot. 
I got confused here because I could not use 10.3.141.1 to access the dashboard before the reboot, not realizing RaspAP had already been enable for the current host instead. 

Then, as billz has been saying probably a million times, I enabled the AP for the WiFi client before a reboot.
However, since the AP is created for the localhost, the DHCP range is incorrect, so I changed it to
DHCP range: 10.3.141.50 to 10.3.141.255 and saved the settings. (I think I might have restarted the dnsmasq service and/or rebooted here).

Finally, at the end of the /etc/dhcpcd.conf file, I used
`
interface uap0
static ip_address=10.3.141.1/24 # changed this from the localhost IP
static domain_name_servers=1.1.1.1 # this was blank before
nohook wpa_supplicant
`
Then I restarted the hostapd service
`sudo systemctl restart hostapd.service`

Then had to wait a bit.

Everything looks to be working to my intent. I can access the WiFi on the Pi and the device connected to the AP! @billz Thanks again! I really appreciate your work, dedication, and assistance to the community!

snehithgit, 2020-05-26:
> Great, thanks guys! I was able to get it working using just the built in WiFi chipset (or additional external WiFi dongle). My WiFi network is 5G and the AP is running on 2.4G (if I understand that [...]

Hi ... I am trying to use wifi ap setting on rpi3 b+ ... I just installed raspbian and raspap ... Without reboot  login to webui and change the setting of wifi ap ... After what to do please clearly guide me .. Thanks in advance

billz, 2020-06-10:
Interested in testing wlan0 + wlan1 support before the next release? See https://github.com/billz/raspap-webgui/issues/580#issuecomment-640458150

closed 2020-06-16 by billz

billz, 2020-06-16:
FAQ updated https://github.com/billz/raspap-webgui/wiki/FAQs#interfaces

MomoSho, 2020-06-19:
@snehithgit you should restart once all the settings are changed

billz, 2023-03-24:
By popular demand, here's a link to our dedicated WiFi repeater walkthrough
https://docs.raspap.com/repeater/