RaspAP/raspap-webgui #337: RaspAP not compatible with Raspbian Buster

CorentinDy, 2019-06-24:
### Subject of the issue

Hi, i've just install raspbian Buster (which is now stable released on rpi), but some packages are no more supported in this release; like "php5-cgi" and maybe some others but can't test .

I know Buster has been released only this morning, but do you have a quick fix for now ? (I didn't tried manual install, but i will if there is no quick fix)
And maybe for later a new release with Buster support ?

I hope there is no big breaking things

### Your environment
* Raspberry Pi hardware (Pi 1)  
* Raspbian version (Buster Lite)
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
* RaspAP Quick Install or Manual setup? Quick Install
* Using default configuration? (Y)
* Other software or services running with RaspAP? No

CorentinDy, 2019-06-24:
As your documentation is really well written, manual installation work perfectly.

billz, 2019-06-24:
Pushed a fix to the Quick Installer to detect Buster. 

As you noted, this was released only this morning. A more complete testing cycle will be performed with Buster shortly. Thanks

closed 2019-06-24 by billz

ldinatale, 2019-06-25:
Hello, I just installed RaspAP on Raspbian Buster (10) performing a semi-manual installation. I downloaded the 'raspapcommon.sh' file and changed the required package php7.0-cgi in php7.1-cgi and then the installation goes well.

billz, 2019-06-25:
Updated installer & uninstaller for php7.1-cgi (Buster). Thanks

commit 13aa0864 (2019-06-24): Updated to detect Raspbian 10.0 (Buster). Resolves #337