RaspAP/raspap-webgui #342: Error saving network configuration to file

m-schmale, 2019-07-05:
### Subject of the issue
When I try to change the network configuration, I get this error: "Error saving network configuration to file"

### Your environment
* Raspberry Pi 3 Model B+  
* Raspbian Buster Lite
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Manual setup
* Using default configuration? Y
* Simultaneous AP and managed mode? N

### Steps to reproduce
I installed RaspAP, changed the default password and then tried to change the network configuration too.

### Expected behavior
The settings should be saved.

### Actual behavior
I received an error message.

Thank you very much in advance :)

billz, 2019-07-06:
Did you set permissions per the manual install steps? `sudo chown -R www-data:www-data /etc/raspap`

m-schmale, 2019-07-07:
> 
> 
> Did you set permissions per the manual install steps? `sudo chown -R www-data:www-data /etc/raspap`

Thanks for the quick reply.

Yes, I followed the manual install steps precisely. I even did it a second time, and I still got the error.
Hopefully, you can localize the error with this information:

ls -l /etc/
`drwxr-xr-x 3 www-data www-data  4096 Jul  7 03:55 raspap`

sudo cat /var/log/syslog | grep hostapd
`Jul  7 03:56:55 raspberrypi hostapd[1014]: Configuration file: /etc/hostapd/hostapd.conf
Jul  7 03:56:55 raspberrypi hostapd[1014]: Could not open configuration file '/etc/hostapd/hostapd.conf' for reading.
Jul  7 03:56:55 raspberrypi hostapd[1014]: Failed to set up interface with /etc/hostapd/hostapd.conf
Jul  7 03:56:55 raspberrypi hostapd[1014]: Failed to initialize interface
Jul  7 03:56:55 raspberrypi systemd[1]: hostapd.service: Control process exited, code=exited, status=1/FAILURE
Jul  7 03:56:55 raspberrypi systemd[1]: hostapd.service: Failed with result 'exit-code'.
Jul  7 03:56:57 raspberrypi systemd[1]: hostapd.service: Service RestartSec=2s expired, scheduling restart.
Jul  7 03:56:57 raspberrypi systemd[1]: hostapd.service: Scheduled restart job, restart counter is at 82.
Jul  7 03:56:57 raspberrypi hostapd[1062]: Configuration file: /etc/hostapd/hostapd.conf
Jul  7 03:56:57 raspberrypi hostapd[1062]: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Jul  7 03:56:57 raspberrypi hostapd[1062]: wlan0: Could not connect to kernel driver
Jul  7 03:56:57 raspberrypi hostapd[1062]: Using interface wlan0 with hwaddr b8:27:eb:5d:42:a1 and ssid "Tor-Pi"
Jul  7 03:56:58 raspberrypi hostapd[1062]: wlan0: interface state COUNTRY_UPDATE->ENABLED
Jul  7 03:56:58 raspberrypi hostapd[1062]: wlan0: AP-ENABLED`

sudo cat /var/log/syslog | grep dnsmasq
`Jul  7 03:42:19 raspberrypi dbus-daemon[421]: Unknown username "dnsmasq" in message bus configuration file
Jul  7 03:42:22 raspberrypi systemd[1]: Starting dnsmasq - A lightweight DHCP and caching DNS server...
Jul  7 03:42:22 raspberrypi dnsmasq[1260]: dnsmasq: syntax check OK.
Jul  7 03:42:22 raspberrypi dnsmasq[1268]: started, version 2.80 cachesize 150
Jul  7 03:42:22 raspberrypi dnsmasq[1268]: DNS service limited to local subnets
Jul  7 03:42:22 raspberrypi dnsmasq[1268]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Jul  7 03:42:22 raspberrypi dnsmasq[1268]: reading /run/dnsmasq/resolv.conf
Jul  7 03:42:22 raspberrypi dnsmasq[1268]: using nameserver 192.168.0.1#53
Jul  7 03:42:22 raspberrypi dnsmasq[1268]: using nameserver <IPv6>
Jul  7 03:42:22 raspberrypi dnsmasq[1268]: read /etc/hosts - 5 addresses
Jul  7 03:42:22 raspberrypi dnsmasq[1269]: Too few arguments.
Jul  7 03:42:22 raspberrypi systemd[1]: Started dnsmasq - A lightweight DHCP and caching DNS server.
Jul  7 03:50:29 raspberrypi systemd[1]: Stopping dnsmasq - A lightweight DHCP and caching DNS server...
Jul  7 03:50:40 raspberrypi systemd[1]: Starting dnsmasq - A lightweight DHCP and caching DNS server...
Jul  7 03:50:40 raspberrypi dnsmasq[573]: dnsmasq: syntax check OK.
Jul  7 03:50:40 raspberrypi dnsmasq[651]: started, version 2.80 cachesize 150
Jul  7 03:50:40 raspberrypi dnsmasq[651]: DNS service limited to local subnets
Jul  7 03:50:40 raspberrypi dnsmasq[651]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Jul  7 03:50:40 raspberrypi dnsmasq[651]: reading /run/dnsmasq/resolv.conf
Jul  7 03:50:40 raspberrypi dnsmasq[651]: using nameserver <IPv6>
Jul  7 03:50:40 raspberrypi dnsmasq[651]: read /etc/hosts - 5 addresses
Jul  7 03:50:41 raspberrypi dnsmasq[652]: Too few arguments.
Jul  7 03:50:41 raspberrypi systemd[1]: Started dnsmasq - A lightweight DHCP and caching DNS server.
Jul  7 03:50:43 raspberrypi dnsmasq[651]: reading /run/dnsmasq/resolv.conf
Jul  7 03:50:43 raspberrypi dnsmasq[651]: using nameserver 192.168.0.1#53
Jul  7 03:50:43 raspberrypi dnsmasq[651]: using nameserver <IPv6>
Jul  7 03:53:32 raspberrypi systemd[1]: Stopping dnsmasq - A lightweight DHCP and caching DNS server...
Jul  7 03:53:43 raspberrypi systemd[1]: Starting dnsmasq - A lightweight DHCP and caching DNS server...
Jul  7 03:53:43 raspberrypi dnsmasq[604]: dnsmasq: syntax check OK.
Jul  7 03:53:43 raspberrypi dnsmasq[656]: started, version 2.80 cachesize 150
Jul  7 03:53:43 raspberrypi dnsmasq[656]: DNS service limited to local subnets
Jul  7 03:53:43 raspberrypi dnsmasq[656]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Jul  7 03:53:43 raspberrypi dnsmasq[656]: reading /run/dnsmasq/resolv.conf
Jul  7 03:53:43 raspberrypi dnsmasq[656]: using nameserver <IPv6>
Jul  7 03:53:43 raspberrypi dnsmasq[656]: read /etc/hosts - 5 addresses
Jul  7 03:53:43 raspberrypi dnsmasq[658]: Too few arguments.
Jul  7 03:53:43 raspberrypi systemd[1]: Started dnsmasq - A lightweight DHCP and caching DNS server.
Jul  7 03:53:45 raspberrypi dnsmasq[656]: reading /run/dnsmasq/resolv.conf
Jul  7 03:53:45 raspberrypi dnsmasq[656]: using nameserver 192.168.0.1#53
Jul  7 03:53:45 raspberrypi dnsmasq[656]: using nameserver <IPv6>
Jul  7 03:53:51 raspberrypi lighttpd[512]: cat: /var/lib/misc/dnsmasq.leases: File or directory not found
[... 28 lines cut ...]
--                         ------      ----
OpenSSH                    ALLOW       Anywhere
[... cut ...]
443/tcp                    ALLOW       Anywhere
OpenSSH (v6)               ALLOW       Anywhere (v6)
22/tcp (v6)                LIMIT       Anywhere (v6)
80/tcp (v6)                ALLOW       Anywhere (v6)
443/tcp (v6)               ALLOW       Anywhere (v6)`

Maybe something is wrong with my configuration?

> Basic settings:
- Interface: wlan0
- SSID: Tor-Pi
- Mode: n
- Channel: 1

Security:
- Security type: WPA2
- Encryption type: CCMP
- PSK: V#UwpGBy!Y5&uCGiLInP

Advanced:
- Country code: Germany

billz, 2019-07-07:
I performed the manual steps on a Pi 3B+ with a fresh install of Buster Lite and could not reproduce this. Your syslog indicates a missing file or incorrect permissions on `hostapd.conf`, which is atypical for a clean install of this project. 

Are you running Tor on this Pi? I noticed your ssid is "Tor-Pi".

m-schmale, 2019-07-07:
That is strange. Maybe I should deactivate the firewall during the installation? And no, I am not running Tor on this Pi (yet).

m-schmale, 2019-07-07:
**stat -c "%a %n" /etc/hostapd/hostapd.conf**
644 /etc/hostapd/hostapd.conf

**ls -l /etc/hostapd**
-rw-r--r-- 1 root root  294 Jul  7 03:56 hostapd.conf

m-schmale, 2019-07-07:
> 
> 
> Your syslog indicates a missing file or incorrect permissions on `hostapd.conf`, which is atypical for a clean install of this project.
>

I have noticed that this file is missing: /etc/raspap/hostapd/servicestart.sh
Is this normal?

billz, 2019-07-07:
> Update: I installed it a third time and this time with disabled ufw. It has not changed anything.

I cannot repro this issue. Please try the Quick Installer on a clean Buster Lite and be sure to follow the project prerequisites _before_ making any configuration changes to your Pi.

closed 2019-07-07 by billz

starmagoo, 2020-01-02:
Hey guys. I got the same error. Can change everything in the web-interface except network configuration (dhcp or static ip). I have installed it manually. I also did the installation a second time, and I still got the error.

SebastienDenooz, 2020-02-04:
I have the same error and it's due to a non existing folder. Here the fix for me:

```
~# mkdir -p /etc/raspap/networking
~# sudo chown -R www-data:www-data /etc/raspap
```