RaspAP/raspap-webgui #343: quick installer broken

pizerox, 2019-07-08:
Hi There,

it seems that the quick installer ist broken.

Last week everything went fine but today wget does not download anything just jumps back to the shell prompt.

I tried the install RaspAP from RaspberryPi's shell prompt:
wget -q https://git.io/voEUQ -O /tmp/raspap && bash /tmp/raspap

Is it possible that the Buster repository was set from "testing" to "stable" in the last days?

Any hint or help is appreciated.

Thanks!

billz, 2019-07-08:
Sure enough. Raspbian Buster was released today, so its status changed to stable.

I've updated the Quick Installer to use `sudo apt update --allow-releaseinfo-change` which resolves this. Thanks.

pizerox, 2019-07-08:
> Closed #343 (https://github.com/billz/raspap-webgui/issues/343).

Ok. Thanks for the quick fix. Tried it again but still the quick installer doesn’t start. Maybe I have to wait until the new release is pushed and delivered.

--  
Henning Lesch
Gesendet mit Sparrow (http://www.sparrowmailapp.com/?sig)

billz, 2019-07-08:
Sure. You can also try `sudo apt-get update --allow-releaseinfo-change` from bash directly and check the output.

pizerox, 2019-07-08:
yes did it already. here is the output - still quick install doesn’t start…

pi@raspberrypi:~ $ sudo apt update --allow-releaseinfo-change  
OK:1 http://archive.raspberrypi.org/debian buster InRelease  
OK:2 http://raspbian.raspberrypi.org/raspbian buster InRelease  
Paketlisten werden gelesen... Fertig  
Abhängigkeitsbaum wird aufgebaut.  
Statusinformationen werden eingelesen.... Fertig
Alle Pakete sind aktuell.
pi@raspberrypi:~ $ wget -q https://git.io/voEUQ -O /tmp/raspap && bash /tmp/raspap
pi@raspberrypi:~ $ sudo apt-get update --allow-releaseinfo-change
OK:1 http://archive.raspberrypi.org/debian buster InRelease  
OK:2 http://raspbian.raspberrypi.org/raspbian buster InRelease  
Paketlisten werden gelesen... Fertig  
pi@raspberrypi:~ $ wget -q https://git.io/voEUQ -O /tmp/raspap && bash /tmp/raspap
pi@raspberrypi:~ $  

--  
Henning Lesch
Gesendet mit Sparrow (http://www.sparrowmailapp.com/?sig)

billz, 2019-07-12:
On Raspbian Buster, `sudo apt update --allow-releaseinfo-change` allows the packages to be updated. You can then run the Quick Installer as normal.

closed 2019-07-12 by billz