RaspAP/raspap-webgui #352: Editing /etc/network/interfaces blows wlan0's config

Macemas, 2019-07-25:
I need to assign a MAC address to device eth1 created when my phone plugged in via USB goes into tether mode. My phone gives a DHCP address only when eth1 has a MAC and therefore I need to edit /etc/network/interfaces and assign a MAC manually for my Samsung (Apparently other phones show up as usb0) to give it a lease and connect.

The RaspAP system is installed on a Pi 3 Model B with the newest Raspbian (7/24/2019), eth0 connected to a 192.168.0.0/24 subnet, grabs DHCP just fine and AP works fine up until /etc/network/interfaces is changed, it's blank and I add:

`
iface eth1
        hwaddress ether B8:26:EB:68:67:97
`
The MAC I'm using is just the wlan0 mac+1, anything works other than all 00

Once these lines are added, and I reboot the system, it shows that wlan0 has no ipv4 address, and sytem dnsmasq status shows that it's receiving DHCP packets but will not give any out due to no IP.
if I run `sudo ifconfig wlan0 10.3.141.1` it fixes the whole issue.

Any ideas how to get around this? just add that last command to my rc.local? I'm hoping to make a redundant mobile network solution, that allows for switching between ethernet and my phone's tether connection.

* Raspberry Pi hardware Pi 3 Model B, 
* Raspbian version: Stretch Lite
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install or Manual setup? Quick install
* Using default configuration? Y(ish)
* Simultaneous AP and managed mode? N
* Other software or services running with RaspAP? None

### Steps to reproduce
Setup RaspAP
edit /etc/networking/interfaces 
wlan0 has no ipv4 address!
### Expected behavior
Setup RaspAP
edit /etc/networking/interfaces to have mac address for eth1
wlan0 has ipv4 address and switches between eth1 if connected and eth0

### Actual behavior
See above

Macemas, 2019-07-25:
I got it to work by making a script startup with the system, that runs
`sudo ifconfig wlan0 10.3.141.1`
as well as configuring eth0 (because that also gets nuked after adding eth1)
here's my /etc/network/interfaces
```
auto eth0
iface eth0 inet dhcp

#Tether
allow-hotplug eth1
iface eth1 inet dhcp
	hwaddress ether B8:26:EB:68:67:97
```

Macemas, 2019-07-25:
The system doesn't automatically switch between eth1 and eth0 so you have to reboot, still works well in this setup. Anyone know how to get it to auto switch routes?

billz, 2019-07-27:
There's an open feature request for similar functionality #267.

Since this doesn't use the default config I can't treat this as a bug, but perhaps your workaround will be useful for others. Thanks

closed 2019-07-27 by billz