RaspAP/raspap-webgui #367: v1.5.1 not accepting configuration changes

feh123, 2019-08-03:
### Subject of the issue
I am trying to configure the hotspot. I change the ssid name and press save settings. Yet the name stays as default. I can try to alter nearly all the hotpot settings - they all default back.

### Your environment
*Pi 3 Model B  
* Raspbian Buster Lite
* Followed the project prerequisites? Y
* Checked the project FAQ? Y
* RaspAP Quick Install
* Using default configuration? Y
* Simultaneous AP and managed mode? Tried both
* Other software or services running with RaspAP? No.

### Steps to reproduce
Just follow intstall guide. I have tried it three times and always get the same problem. If I revert to v1.5 everything works.

### Expected behavior
I should be able to configure the hotspot.

### Actual behavior
I cannot change hotspot configuration.

Blimpyway, 2019-08-03:
Yup I can confirm same problem.

Blimpyway, 2019-08-03:
So the problem happens on Pi Zero W in both buster and stretch . 
Also on Pi Zero with buster and  an Edimax usb wifi dongle
` (Bus 001 Device 003: ID 7392:7711 Edimax Technology Co., Ltd EW-7711UTn nLite Wireless Adapter [Ralink RT2870])
`

Here-s an output of tail -f /var/log/lighttpd/error.log: 

`
2019-08-04 00:57:19: (mod_fastcgi.c.2543) FastCGI-stderr: Attempting to set hostapd config with wpa='1', wpa_pairwise='TKIP' and hw_mode='g'
2019-08-04 00:57:19: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Notice:  Undefined variable: countries_max11channels in /var/www/html/includes/hostapd.php on line 136
2019-08-04 00:57:19: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Warning:  in_array() expects parameter 2 to be array, null given in /var/www/html/includes/hostapd.php on line 136
2019-08-04 00:57:19: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Notice:  Undefined index: LogEnable in /var/www/html/includes/hostapd.php on line 179
2019-08-04 00:57:19: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Notice:  Undefined index: WifiAPEnable in /var/www/html/includes/hostapd.php on line 196
2019-08-04 00:57:19: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Notice:  Undefined index: LogEnable in /var/www/html/includes/hostapd.php on line 210
2019-08-04 00:57:19: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Notice:  Undefined index: ignore_broadcast_ssid in /var/www/html/includes/hostapd.php on line 224
2019-08-04 00:57:19: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Notice:  Undefined index: ignore_broadcast_ssid in /var/www/html/includes/hostapd.php on line 224
             
`

Also I noticed

feh123, 2019-08-04:
I also tried a Pi zero WH with Buster lite and it failed to configure - as above. I tried a modmypi RT5370 designed for RPI zero has the second wifi - raspap could create a wlan1 up but no hotspot. [This is a different issue but I thought I would mention it].

billz, 2019-08-04:
Thanks for the debug info. I'll look into this tomorrow, lots of open PR's to test/review also

billz, 2019-08-05:
We check hostapd.conf inputs for bad data and log an error if expected values aren't present. A recent update for descriptive labels in `$arr80211Standard` causes this check to fail. 
https://github.com/billz/raspap-webgui/blob/02c3703392963938edebd9f73284ae840b8ecbff/includes/hostapd.php#L525

I will push a fix shortly.

closed 2019-08-05 by billz

glaszig, 2019-08-05:
my fault. sorry :(

billz, 2019-08-05:
@glaszig not at all. I should have tested more carefully 😉

commit c0cfcfa1 (2019-08-05): Bugfix, resolves #367