RaspAP/raspap-webgui #368: WiFi scanning  doesn't work for quick install on raspbian buster running on raspberry pi 4b

3rdimagiccircle, 2019-08-05:
WiFi scan isn't working
RaspAp successfully  creates  AP on raspbian buster running on raspberry pi 4b and I could connect to  it but WiFi scanning in WiFi client doesn't work.

I have tried both quick and manual installation methods but WiFi scanning didn't work for both the cases. 

I had faced similar  issues on raspberry pi 3b with raspbian stretch lite for quick install but it got sorted out after reinstalling raspap using manual installation method.

My environment
* Raspberry Pi 4b
* Raspbian version : buster lite
* Followed the project prerequisites? (Y)
* Checked the project FAQ? (Y)
*Installation type- didn't work for both quick install and manual install.
* Using default configuration? (Y)
* Simultaneous AP and managed mode? (N)
* Other software or services running with RaspAP? - N

### Steps to reproduce
Quick install raspap on raspbian buster running on  raspberry pi 4b. onboard wifi adapter is acting as AP and an external wifi adapter is used as station.
Onboard wifi adaptor is named wap0 by hard coding its mac address in udev settings.

### Expected behavior
External adapter which gets detected as wlan0 must be able to scan for wifi access points and connect to whichever user chooses.

### Actual behavior
Scanning doesn't work, 
If I manually insert  AP name in wpa_supplicant.conf, name of the AP gets displayed on clients page but if I delete the AP on the client page raspap fails to restart the interface.

glaszig, 2019-08-05:
please, post the output of `iw list`. thank you.

hetsen, 2019-08-06:
`root@raspberrypi:/home/pi# iw list
Wiphy phy0
        max # scan SSIDs: 10
        max scan IEs length: 2048 bytes
        max # sched scan SSIDs: 16
        max # match sets: 16
        max # scan plans: 1
        max scan plan interval: 508
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports roaming.
        Device supports T-DLS.
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CMAC (00-0f-ac:6)
        Available Antennas: TX 0 RX 0
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * P2P-client
                 * P2P-GO
                 * P2P-device
        Band 1:
                Capabilities: 0x1062
                        HT20/HT40
                        Static SM Power Save
                        RX HT20 SGI
                        RX HT40 SGI
                        No RX STBC
                        Max AMSDU length: 3839 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 16 usec (0x07)
                HT TX/RX MCS rate indexes supported: 0-7
                Bitrates (non-HT):
                        * 1.0 Mbps
                        * 2.0 Mbps (short preamble supported)
                        * 5.5 Mbps (short preamble supported)
                        * 11.0 Mbps (short preamble supported)
                        * 6.0 Mbps
                        * 9.0 Mbps
                        * 12.0 Mbps
                        * 18.0 Mbps
                        * 24.0 Mbps
                        * 36.0 Mbps
                        * 48.0 Mbps
                        * 54.0 Mbps
                Frequencies:
                        * 2412 MHz [1] (20.0 dBm)
                        * 2417 MHz [2] (20.0 dBm)
                        * 2422 MHz [3] (20.0 dBm)
                        * 2427 MHz [4] (20.0 dBm)
                        * 2432 MHz [5] (20.0 dBm)
                        * 2437 MHz [6] (20.0 dBm)
                        * 2442 MHz [7] (20.0 dBm)
                        * 2447 MHz [8] (20.0 dBm)
                        * 2452 MHz [9] (20.0 dBm)
                        * 2457 MHz [10] (20.0 dBm)
                        * 2462 MHz [11] (20.0 dBm)
                        * 2467 MHz [12] (20.0 dBm)
                        * 2472 MHz [13] (20.0 dBm)
                        * 2484 MHz [14] (disabled)
        Band 2:
                Capabilities: 0x1062
[... 95 lines cut ...]
        Supported TX frame types:
                 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
        Supported RX frame types:
                 * managed: 0x40 0xd0
                 * P2P-client: 0x40 0xd0
                 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * P2P-device: 0x40 0xd0
        software interface modes (can always be added):
        valid interface combinations:
                 * #{ managed } <= 1, #{ P2P-device } <= 1, #{ P2P-client, P2P-GO } <= 1,
                   total <= 3, #channels <= 2
                 * #{ managed } <= 1, #{ AP } <= 1, #{ P2P-client } <= 1, #{ P2P-device } <= 1,
                   total <= 4, #channels <= 1
        Device supports scan flush.
        Device supports randomizing MAC-addr in sched scans.
        Supported extended features:
                * [ 4WAY_HANDSHAKE_STA_PSK ]: 4-way handshake with PSK in station mode
                * [ 4WAY_HANDSHAKE_STA_1X ]: 4-way handshake with 802.1X in station mode
root@raspberrypi:/home/pi#
`

I got the same issue, this is what it looks like for me.

glaszig, 2019-08-06:
alright. i need more; the outputs of the following commands. and please wrap them in code blocks (```).

`ifconfig -a`

`sudo wpa_cli -i wlan0 scan`

`sudo wpa_cli -i wlan0 scan_results`

`sudo iwlist wlan0 scan`

thanks.

3rdimagiccircle, 2019-08-06:
Hi following is the result of iw list

Wiphy phy0
        max # scan SSIDs: 4
        max scan IEs length: 2257 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CCMP-256 (00-0f-ac:10)
                * GCMP-128 (00-0f-ac:8)
                * GCMP-256 (00-0f-ac:9)
                * CMAC (00-0f-ac:6)
                * CMAC-256 (00-0f-ac:13)
                * GMAC-128 (00-0f-ac:11)
                * GMAC-256 (00-0f-ac:12)
        Available Antennas: TX 0 RX 0
        Supported interface modes:
                 * managed
                 * monitor
        Band 1:
                Capabilities: 0x17e
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 2 usec (0x04)
                HT TX/RX MCS rate indexes supported: 0-7, 32
                Bitrates (non-HT):
                        * 1.0 Mbps (short preamble supported)
                        * 2.0 Mbps (short preamble supported)
                        * 5.5 Mbps (short preamble supported)
                        * 11.0 Mbps (short preamble supported)
                        * 6.0 Mbps
                        * 9.0 Mbps
                        * 12.0 Mbps
                        * 18.0 Mbps
                        * 24.0 Mbps
                        * 36.0 Mbps
                        * 48.0 Mbps
                        * 54.0 Mbps
                Frequencies:
                        * 2412 MHz [1] (20.0 dBm)
                        * 2417 MHz [2] (20.0 dBm)
                        * 2422 MHz [3] (20.0 dBm)
                        * 2427 MHz [4] (20.0 dBm)
                        * 2432 MHz [5] (20.0 dBm)
                        * 2437 MHz [6] (20.0 dBm)
                        * 2442 MHz [7] (20.0 dBm)
                        * 2447 MHz [8] (20.0 dBm)
                        * 2452 MHz [9] (20.0 dBm)
                        * 2457 MHz [10] (20.0 dBm)
                        * 2462 MHz [11] (20.0 dBm)
                        * 2467 MHz [12] (20.0 dBm)
                        * 2472 MHz [13] (20.0 dBm)
                        * 2484 MHz [14] (disabled)
[... 234 lines cut ...]
                 * disconnect
                 * crit_protocol_start
                 * crit_protocol_stop
                 * update_connect_params
        Supported TX frame types:
                 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
        Supported RX frame types:
                 * managed: 0x40 0xd0
                 * P2P-client: 0x40 0xd0
                 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * P2P-device: 0x40 0xd0
        software interface modes (can always be added):
        valid interface combinations:
                 * #{ managed } <= 1, #{ P2P-device } <= 1, #{ P2P-client, P2P-GO } <= 1,
                   total <= 3, #channels <= 2
                 * #{ managed } <= 1, #{ AP } <= 1, #{ P2P-client } <= 1, #{ P2P-device } <= 1,
                   total <= 4, #channels <= 1
        Device supports scan flush.
        Device supports randomizing MAC-addr in sched scans.
        Supported extended features:
                * [ 4WAY_HANDSHAKE_STA_PSK ]: 4-way handshake with PSK in station mode
                * [ 4WAY_HANDSHAKE_STA_1X ]: 4-way handshake with 802.1X in station mode

3rdimagiccircle, 2019-08-06:
Following is the result of  ifconfig -a
> eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.173.20  netmask 255.255.255.0  broadcast 192.168.173.255
        inet6 fe80::dea6:32ff:fe05:63d3  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:05:63:d3  txqueuelen 1000  (Ethernet)
        RX packets 3819  bytes 443960 (433.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2313  bytes 295942 (289.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

> lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 51496  bytes 3425232 (3.2 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 51496  bytes 3425232 (3.2 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

> wap0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.3.141.1  netmask 255.255.255.0  broadcast 10.3.141.255
        inet6 fe80::7345:a9e0:7d1b:96ce  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:05:63:d4  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 40  bytes 5360 (5.2 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

> wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 00:e1:29:00:c0:2d  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

sudo wpa_cli -i wlan0 scan gave following rspponse
> ok

sudo wpa_cli -i wlan0 scan_results 

> Showed  bssid / frequency / signal level / flags / ssid  of available access points.

sudo iwlist wlan0 scan gave following result

>wlan0     Scan completed :
          Cell 01 - Address: C4:A8:1D:3F:78:7A
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
                    Quality=39/70  Signal level=-71 dBm
                    Encryption key:on
                    ESSID:"BSNL99"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 6 Mb/s
                              9 Mb/s; 12 Mb/s; 18 Mb/s
                    Bit Rates:24 Mb/s; 36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=00000067af55bd83
                    Extra: Last beacon: 1380ms ago
                    IE: Unknown: 000642534E4C3939
                    IE: Unknown: 010882848B960C121824
                    IE: Unknown: 030101
                    IE: Unknown: 2A0100
                    IE: Unknown: 32043048606C
                    IE: Unknown: 2D1A6E1817FF00000000000000000000000000000000000000000000
                    IE: Unknown: 3D1601050000000000000000000000000000000000000000
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : TKIP
                        Pairwise Ciphers (2) : TKIP CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown: DD180050F2020101000003A4000027A4000042435E0062322F00
                    IE: Unknown: DD7C0050F204104A0001101044000102103B0001031047001063041253101920061228C4A81D3F787A10210006442D4C696E6B1023000752544C383637311024000D45562D323030362D30372D32371042000C6334613831643366373837311054000800060050F20400011011000944534C2D3237333055100800020086
[... 66 lines cut ...]
                    Bit Rates:24 Mb/s; 36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000010d92f89808
                    Extra: Last beacon: 940ms ago
                    IE: Unknown: 00077072617665656E
                    IE: Unknown: 010882848B960C121824
                    IE: Unknown: 030105
                    IE: Unknown: 2A0100
                    IE: Unknown: 32043048606C
                    IE: Unknown: 2D1AEF191FFFFF000000000000000000000000000000001804810800
                    IE: Unknown: 3D1605070000000000000000000000000000000000000000
                    IE: Unknown: 4A0E14000A00B400C800140005001900
                    IE: WPA Version 1
                        Group Cipher : TKIP
                        Pairwise Ciphers (2) : TKIP CCMP
                        Authentication Suites (1) : PSK
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : TKIP
                        Pairwise Ciphers (2) : TKIP CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown: DD180050F2020101800003A4000027A4000042435E0062322F00
                    IE: Unknown: DD1E00904C33EF191FFFFF000000000000000000000000000000001804810800
                    IE: Unknown: DD1A00904C3405070000000000000000000000000000000000000000
                    IE: Unknown: DD0600E04C020160
                    IE: Unknown: 7F080100000000000000

hetsen, 2019-08-07:
ipconfig -a

```
root@raspberrypi:/home/pi# ifconfig -a
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.2.221  netmask 255.255.255.0  broadcast 192.168.2.255
        inet6 fe80::77ee:808c:297:d687  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:05:7d:99  txqueuelen 1000  (Ethernet)
        RX packets 1086  bytes 122170 (119.3 KiB)
        RX errors 0  dropped 3  overruns 0  frame 0
        TX packets 392  bytes 75193 (73.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 16  bytes 1342 (1.3 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 16  bytes 1342 (1.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.3.141.1  netmask 255.255.255.0  broadcast 10.3.141.255
        inet6 fe80::ea11:1654:7b37:fc8f  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:05:7d:9a  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 41  bytes 5696 (5.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

root@raspberrypi:/home/pi#
```

sudo wpa_cli -i wlan0

```
root@raspberrypi:/home/pi# sudo wpa_cli -i wlan0
wpa_cli v2.8-devel
Copyright (c) 2004-2019, Jouni Malinen <j@w1.fi> and contributors

This software may be distributed under the terms of the BSD license.
See README for more details.

Interactive mode

Could not connect to wpa_supplicant: wlan0 - re-trying
```

sudo wpa_cli -i wlan0 scan_results

```
root@raspberrypi:/home/pi# sudo wpa_cli -i wlan0 scan_results
Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory
```

 sudo iwlist wlan0 scan

```
root@raspberrypi:/home/pi# sudo iwlist wlan0 scan
wlan0     Scan completed :
          Cell 01 - Address: 84:80:2D:43:3D:E8
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
                    Quality=37/70  Signal level=-73 dBm
                    Encryption key:on
                    ESSID:"MAMINT_NYK"
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 18 Mb/s; 24 Mb/s
                              36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
[... 260 lines cut ...]
                              36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 10ms ago
                    IE: Unknown: 000D54616E6B62617220556E694669
                    IE: Unknown: 01088C129824B048606C
                    IE: Unknown: 03012C
                    IE: Unknown: 050400010000
                    IE: Unknown: 200103
                    IE: Unknown: 2D1AEF091BFFFF000000000000000000000100000000000000000000
                    IE: Unknown: 3D162C0D0C00000000000000000000000000000000000000
                    IE: Unknown: 7F080000000000000040
                    IE: Unknown: BF0CB2018033FAFF0000FAFF0000
                    IE: Unknown: C005000000FCFF
                    IE: Unknown: C30402D8D8D8
                    IE: Unknown: DD180050F2020101000003A4000027A4000042435E0062322F00
                    IE: Unknown: DD0900037F01010000FF7F
                    IE: Unknown: DD1300156D00010100010217E58106788A20508705
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK

root@raspberrypi:/home/pi#
```

hetsen, 2019-08-07:
> ipconfig -a
>
> ```
> root@raspberrypi:/home/pi# ifconfig -a
> eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
>         inet 192.168.2.221  netmask 255.255.255.0  broadcast 192.168.2.255 [...]

after some research I noticed that it has to do with hostapd, Installing an older version resulted in me being able to search for wifi but RaspAP won't be able to enable the AP.

glaszig, 2019-08-07:
> Installing an older version

would you mind sharing the version you had and the version you're using now?

hetsen, 2019-08-08:
> > Installing an older version
> 
> would you mind sharing the version you had and the version you're using now?
> 
> ```
> $ hostapd -v
> hostapd v2.8-devel
> User space daemon for IEEE 802.11 AP management,
> IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
> Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
> ```

The 

 hostapd (2:2.4-1+deb9u4) package made it possible to scan for networks but could not activate the AP.

glaszig, 2019-08-09:
are both of you sure you ran the latest version of wpa_supplicant?

@hetsen your `wpa_cli -i wlan0` (which should have been `wpa_cli -i wlan0 scan`) command said it couldn‘t connect to wpa_supplicant. so it was not running. i just ask myself why...

do you both have configured the correct country setting in wpa_supplicant.conf?

afaik that‘s not set by default.

more details: https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md

3rdimagiccircle, 2019-08-10:
Hi, rpi is running latest version of wpa_supplicant and I have corrected country code in it, that doesn't seems to be reason for raspap not scanning.

billz, 2019-09-09:
@3rdimagiccircle please post the output of:
`sudo wpa_supplicant -iwlan0 -c/etc/wpa_supplicant/wpa_supplicant.conf`

You might see the message "ctrl_iface exists and seems to be in use", in which case follow the recommendation and delete '/var/run/wpa_supplicant/wlan0' manually.

billz, 2019-09-10:
These errors:

> Could not connect to wpa_supplicant: wlan0 - re-trying
> Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory

...almost always indicate a problem with the socket used to communicate with wpa_supplicant. Reinitialize it with:
```
sudo wpa_supplicant -B -Dnl80211,wext -c/etc/wpa_supplicant/wpa_supplicant.conf -iwlan0
```
You should then be able to perform scans as expected.

Please be aware that this does not necessarily point to a problem with RaspAP. This project simply uses `wpa_cli -i wlan0 scan` to populate a list of available wireless networks. If you can't execute this from the shell, neither can the web UI.

billz, 2019-09-15:
Marking this as resolved. If other users encounter this problem, please refer to the post above.

closed 2019-09-15 by billz

billz, 2019-09-25:
I've updated the FAQ to address this. Meanwhile, I'll investigate possible root causes as it appears this is not a global phenomenon.

musikele, 2021-06-01:
Hey @billz , 
I am stuck here: 

```
pi@raspberrypi:~ $ wpa_cli -i wlan0 scan
Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory
```

and: 

```
pi@raspberrypi:~ $ sudo wpa_supplicant -B -Dnl80211,wext -c/etc/wpa_supplicant/wpa_supplicant.conf -iwlan0
Successfully initialized wpa_supplicant
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
Connection to 10.3.141.1 closed by remote host.
```

basically after reinitializing wpa_supplicant raspberry blocks and I get kicked out. Also wifi signal is interrupted. How can I help debugging the problem?

billz, 2021-06-01:
@musikele use `-dd` with wpa_supplicant for verbose debug info. you can also log output to a file instead of stdout, which is useful if you're getting kicked off.

Quite often, though not always, the above output occurs when another wpa_supplicant process is running, or wasn't terminated cleanly. In this event, use `sudo killall wpa_supplicant`. best of luck

musikele, 2021-06-03:
@billz here's the output: 

```
pi@raspberrypi:~ $ sudo wpa_supplicant -dd -B -Dnl80211,wext -c /etc/wpa_supplicant/wpa_supplicant.conf -iwlan0
wpa_supplicant v2.8-devel
random: getrandom() support available
Successfully initialized wpa_supplicant
Initializing interface 'wlan0' conf '/etc/wpa_supplicant/wpa_supplicant.conf' driver 'nl80211,wext' ctrl_interface 'N/A' bridge 'N/A'
Configuration file '/etc/wpa_supplicant/wpa_supplicant.conf' -> '/etc/wpa_supplicant/wpa_supplicant.conf'
Reading configuration file '/etc/wpa_supplicant/wpa_supplicant.conf'
ctrl_interface='DIR=/var/run/wpa_supplicant GROUP=netdev'
update_config=1
Line: 3 - start of a new network block
ssid - hexdump_ascii(len=12):
     57 49 46 49 20 54 52 49 4e 49 54 59               WIFI TRINITY
PSK - hexdump(len=32): [REMOVED]
Priority group 0
   id=0 ssid='WIFI TRINITY'
nl80211: Using driver-based roaming
nl80211: TDLS supported
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:6
nl80211: Using driver-based off-channel TX
nl80211: Supported vendor command: vendor_id=0x1018 subcmd=1
nl80211: Use separate P2P group interface (driver advertised support)
nl80211: Enable multi-channel concurrent (driver advertised support)
nl80211: use P2P_DEVICE support
nl80211: interface wlan0 in phy phy0
nl80211: Set mode ifindex 3 iftype 3 (AP)
nl80211: Setup AP(wlan0) - device_ap_sme=1 use_monitor=0
nl80211: Subscribe to mgmt frames with AP handle 0x316ea0 (device SME)
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x316ea0 match=04
nl80211: kernel reports: Match already configured
nl80211: Register frame command failed (type=208): ret=-114 (Operation already in progress)
nl80211: Register frame match - hexdump(len=1): 04
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x316ea0 match=0501
nl80211: kernel reports: Match already configured
nl80211: Register frame command failed (type=208): ret=-114 (Operation already in progress)
nl80211: Register frame match - hexdump(len=2): 05 01
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x316ea0 match=0503
nl80211: kernel reports: Match already configured
nl80211: Register frame command failed (type=208): ret=-114 (Operation already in progress)
nl80211: Register frame match - hexdump(len=2): 05 03
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x316ea0 match=0504
nl80211: kernel reports: Match already configured
nl80211: Register frame command failed (type=208): ret=-114 (Operation already in progress)
nl80211: Register frame match - hexdump(len=2): 05 04
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x316ea0 match=06
nl80211: kernel reports: Match already configured
nl80211: Register frame command failed (type=208): ret=-114 (Operation already in progress)
nl80211: Register frame match - hexdump(len=1): 06
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x316ea0 match=08
nl80211: kernel reports: Match already configured
nl80211: Register frame command failed (type=208): ret=-114 (Operation already in progress)
nl80211: Register frame match - hexdump(len=1): 08
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x316ea0 match=09
nl80211: kernel reports: Match already configured
nl80211: Register frame command failed (type=208): ret=-114 (Operation already in progress)
nl80211: Register frame match - hexdump(len=1): 09
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x316ea0 match=0a
nl80211: kernel reports: Match already configured
nl80211: Register frame command failed (type=208): ret=-114 (Operation already in progress)
nl80211: Register frame match - hexdump(len=1): 0a
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x316ea0 match=11
nl80211: kernel reports: Match already configured
nl80211: Register frame command failed (type=208): ret=-114 (Operation already in progress)
nl80211: Register frame match - hexdump(len=1): 11
[... 13 lines cut ...]
nl80211: Register frame match - hexdump(len=0): [NULL]
nl80211: Failed to enable Probe Request frame reporting in AP mode
rfkill: initial event: idx=0 type=1 op=0 soft=0 hard=0
netlink: Operstate: ifindex=3 linkmode=1 (userspace-control), operstate=5 (IF_OPER_DORMANT)
Add interface wlan0 to a new radio phy0
nl80211: Regulatory information - country=00
nl80211: 2402-2472 @ 40 MHz 20 mBm
nl80211: 2457-2482 @ 20 MHz 20 mBm (no IR)
nl80211: 2474-2494 @ 20 MHz 20 mBm (no OFDM) (no IR)
nl80211: 5170-5250 @ 80 MHz 20 mBm (no IR)
nl80211: 5250-5330 @ 80 MHz 20 mBm (DFS) (no IR)
nl80211: 5490-5730 @ 160 MHz 20 mBm (DFS) (no IR)
nl80211: 5735-5835 @ 80 MHz 20 mBm (no IR)
nl80211: 57240-63720 @ 2160 MHz 0 mBm
nl80211: Added 802.11b mode based on 802.11g information
nl80211: Mode IEEE 802.11g: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 2484
nl80211: Mode IEEE 802.11a: 5170 5180 5190 5200 5210 5220 5230 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825
nl80211: Mode IEEE 802.11b: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 2484
Connection to 10.3.141.1 closed by remote host.
Connection to 10.3.141.1 closed.
```

Most importantly, raspap stopped to operate. Even though my mac is still connected to the network, I cannot navigate, connect via ssh, or connect to web admin. 

I've also tried to kill wpa_supplicant before reinitializing, same effect. In the end I have to shutdown and restart my raspberry. But still I cannot scan for networks and/or connect.

billz, 2021-06-03:
> nl80211: Regulatory information - country=00

It appears you have not set a country code.

> Do you think I should create a separate ticket?

If you can't execute this from the shell, neither can this project. 
RaspAP abstracts a lot of this for you, but it isn't magic.

mw2323, 2022-03-18:
I am having the exact same issue stated above. Clean install on a Pi4 running Buster Lite. Followed all instructions to a tee. Set my wifi country correctly. Raspap is able to create a hotspot, but not able to scan for existing wifi. I would really like to get this working. I had this same issue years ago and abandoned the project entirely until now.

I have tried several times a brand new clean installation.
I am getting the following, same as musikele:

```
pi@raspberrypi:~ $ wpa_cli -i wlan0 scan
Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory
```

```
pi@raspberrypi:~ $ sudo wpa_supplicant -B -Dnl80211,wext -c/etc/wpa_supplicant/wpa_supplicant.conf -iwlan0
Successfully initialized wpa_supplicant
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
Connection to 10.3.141.1 closed by remote host.
```

When I run `sudo killall wpa_supplicant`, my hotspot stops functioning entirely and I have to restart the Pi and start over.

Do I need a separate network interface for the AP and another one to connect to an existing network to provide internet?

Again, I have set a country code in Raspi-config (but after installing RaspAP via quick installer, Raspi-config can no longer communicate with wpa_supplicant either. Something is broken.

billz, 2022-03-18:
> Do I need a separate network interface for the AP and another one to connect to an existing network to provide internet?

Yes.

musikele, 2022-03-21:
@billz 
So, to be 100% clear: in order to wifi-scan AND connect to a new network we need to buy a USB wifi dongle. Am I right? 

If I am, this piece of info was not written anywhere, and it's the first time I read this - and I went through documentation many times...

billz, 2022-03-21:
@musikele only if all the above workarounds do not solve it for you. this project provides a front-end for wpa_supplicant only. 
if it fails on the command line do not expect different results from the UI.