RaspAP/raspap-webgui #373: can't connect with Ubuntu or raspberry

laurent1981be, 2019-08-08:
### Before submitting an issue

Please read this first https://github.com/billz/raspap-webgui/wiki/Reporting-issues. 

### Subject of the issue
I installed raspap on a rpi2 on a raspbian freshly burned but i can't connect with different devices :
- Laptop Ubuntu : KO
- laptop win10 : OK
- Android : ok
- iPhone : KO (tryed 1 time...)
- raspberry (zero w) : KO

does anyone have the same issue? 

My project is to connect my Zero W with motion eye os  and camera by WiFi to my Pi2 who runs motion eye detection.

### Your environment
* Raspberry Pi hardware Pi2
* Raspbian version : Stretch desktop, buster desktop
* Followed the project prerequisites? Yes
* Checked the project FAQ? Yes, try changing to CCMP but no changes
* RaspAP Quick Install or Manual setup? Quick install
* Using default configuration? Yes
* Simultaneous AP and managed mode? No
* Other software or services running with RaspAP? Motion eye. 

### Steps to reproduce
Burn raspbian
Update and upgrade
Install raspap quickly (without motion eye)
Let the defaults settings and try connect wifi

### Expected behavior
Can connect with any devices OS

### Actual behavior
Connect on win10 or Android
Don't connect with Ubuntu, raspbian or IOS.

billz, 2019-08-08:
> I installed raspap on a rpi2

The RPi 2 does not have an integrated wireless LAN chipset.

laurent1981be, 2019-08-08:
@billz, thanks for answer, but i got a wifi adapter on a USB port and the wifi works well when it's connected on my router.
It's the reason why i can connect with my android phone if the AP is on.
Do you think if the wireless chipset is not integrated, the AP doesn't work fine ?

billz, 2019-08-08:
> i got a wifi adapter on a USB port

OK, your original report didn't mention an external adapter. Wifi dongles vary in terms of hardware capabilities and driver support. If you can connect Android and Windows clients we can assume it at least supports AP mode. Edit your hostapd.conf with `sudo nano /etc/hostapd/hostapd.conf` and add the following to the end:

```
logger_syslog=-1
logger_syslog_level=1
```

This will output debugging info to the syslog. Save the config and restart hostapd. Monitor the service with `sudo tail -f /var/log/syslog` and try connecting iOS and Ubuntu clients. Compare the output to clients that are able to connect to the AP.

commit 99b0c04e (2019-08-12): Bugfix in sudoers. Resolves #373
closed 2019-08-12 by billz